CR6Community/Marlin #84: Provisional release notes for CR-6 community firmware release 5 beta

Sebazzz, 2020-12-23:
First of all I want to thank everyone who has been [testing the alpha 4 release of the community firmware](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-4-alpha). All the testing and feedback has made this firmware better and better. This is *truly* a community firmware.

In the period between then and now, my Creality CR-6 has died and from its ashes it became alive once again with a [BigTreeTech SKR CR-6](https://damsteen.nl/blog/2020/11/25/how-to-btt-skr-cr6-installation) board. A while later I also received a new Creality CR-6 SE [with the v4.5.3](https://www.reddit.com/r/CR6/comments/k97r5d/creality_cr6_se_new_motherboard_v453/) motherboard. It took a while for that one to arrive, then a while for the firmware source code to be released, so it has caused some delays to get real work done. Yet, some users were able to test some fixes I was able to do without a operational printer.

*TL;DR: CR-6 MAX support, better Octoprint support, BigTreeTech SKR CR-6 support, native and improved filament sensor support, many stability and ABL accuracy improvements, rebased on latest Marlin, and enabled the watchdog safety feature*

## Table of contents

- Special thanks
- Other news
- Major features (all firmware editions)
- Detailed changes
- Comparison with Creality stock firmware
- Known issues
- Flashing instructions
- FAQ
- Reporting issues
- Contributing

## Special thanks to...

- @grobux for designing a completely new user interface for Creality stock touch screen
- @Thinkersbluff, @Beeble2695 and @InsanityAutomation as a sparring partner for various firmware and hardware mechanics of the CR-6
- @InsanityAutomation for further refining the bed leveling code and submitting some things upstream
- @Nushio for putting up testing protocols and doing an insane amount of stress testing on the firmware
- @dabbodev for contributing configuration changes to support for the Creality CR-6 MAX
- The various testers and contributors to the firmware, either through testing, feedback, as a sounding board to the questionnaire we sent out earlier, or even donations!

If you like to send some love to the developers, [you can find more information here](https://github.com/CR6Community/Marlin#credits).

## Other news

- @InsanityAutomation is using the CR-6 community firmware and rebranding it for use on machines sold [by Tiny Machines](https://www.tinymachines3d.com/)
- Some auto bed leveling features have already been merged to upstream Marlin

## Major features across firmware editions

_Note: This is *still* a beta formally - because this is based on a unreleased version of Marlin. However, thanks to the amount of effort that has gone into it, I'm confident it should be a pretty stable release!_

**Important safety feature: We now enable the watchdog for STM32F1 processors - if Marlin would freeze from some reason (like due to SD card issues) the printer is automatically reset and the heaters are disabled - this feature is not enabled on the stock Creality or BTT firmware**

### Hardware support
- Support for Creality v4.5.2 and the new v4.5.3 motherboards. See the [FAQ](#faq) section to know which board you have.
- Support for [the BigTreeTech SKR CR-6 board](https://damsteen.nl/blog/2020/11/25/how-to-btt-skr-cr6-installation), both with BTT TFT and stock Creality DWIN TFT
- Support for the Creality CR-6 MAX

### Reliability
- Rebased on latest Marlin bugfix-2.0.x sources which will eventually become Marlin v2.0.7.3
- Increase bed leveling and homing (Z-offset) accuracy
   - We do this by preheating before probing and turning off heaters when probing so electrical interference when reading the strain gauge is minimized
- Further tinkering with the movement settings for optimum performance

##  Detailed Changes 

- Both homing and leveling preheat to decrease the effect of temperature differences and filament stuck to the nozzle causing issues
- Heaters are turned off when the actual probe is done to decrease electrical noise, which then causes Z-offset and bed leveling inaccuracies
- Improve reliability of probing and leveling (issue #71, issue #67, issue #49, issue #42, issue #41, issue #38)
- Increase serial buffer size to improve reliability when printing a lot of vases via a host like Octoprint
- Automatically enable bed leveling after homing - even if you haven't called `M420 S1` before
- Enable `G29` bed leveling mesh validation tool
- Enable PID tuning for the heated bed
- Enable hotend time-out. This feature automatically disables the hot-end to prevent charring and jams. Currently set to 5 minutes.
- Enable G2/G3 arc by default - useful [for those using Arc Welder](https://plugins.octoprint.org/plugins/arc_welder/)
- Enable host action commands and host prompt support
- Enable probe accuracy test M48

### Changes specific when using Creality v4.5.2 and v4.5.3 motherboards

- Improve reliability for SD card prints freezing - especially if using v4.5.2 boards (issue #62)
- Improve reliability in vase mode, prevent blobs when printing vases
[... 168 lines cut ...]

- Verify that the issues are not in your hardware. We've seen issues that were reported as firmware issues, but were actually hardware issues that "just" happened.
    - Check you have a good SD card. Do **not** use the Creality SD card that came with your printer. It will fail sooner or later and result in failures. Try a different SD card.
    - Leveling: try it again a second time and try using the stock cable clips.

- Verify that the issues are not in your slicer
   - Cura 4.7.x is having issues
   - Cura 4.8 is still giving some issues for some people. Try playing with the maximum resolution and maximum deviation settings in your slicing profile.

## Feature requests

Feature requests are welcome! Let us know in [the issue tracker](https://github.com/CR6Community/Marlin/issues/)!

## Contibuting

Your contributions are very welcome! 

- Please open [an issue in the issue tracker first](https://github.com/CR6Community/Marlin/issues/) to discuss what you like to contribute. This is especially important if it involves the touch screen because due to technical limitations only one person can work on the touch screen firmware at the same time.
- Issues marked "[good first issue](https://github.com/CR6Community/Marlin/issues?q=is%3Aopen+is%3Aissue+label%3A%22good+first+issue%22)" or "[help wanted](https://github.com/CR6Community/Marlin/issues?q=is%3Aopen+is%3Aissue+label%3A%22help+wanted%22)" can are a good issue to start with!

## Some final notes

- Remember that this is a community project, we're not paid (definitely not by Creality!), we have a full time job, doing this in our spare time, and we're just 3d printing enthusiasts who like to have a better Creality CR-6 experience for everyone.
- We're active in the [CR-6 community discord](https://discord.gg/RKrxYy3Q9N) - come say hello!
- You can get the files in the "assets" section below. Note that if you like to compile the firmware yourself [you can best pick the correct branch from Github instead](https://github.com/CR6Community/Marlin/#development-and-compile-it-yourself)

Thinkersbluff, 2020-12-24:
“ Enable G29 bed leveling mesh validation tool” may be typo?  
Mesh validation tool is G26.

Sebazzz, 2020-12-24:
Fixed it. Thanks for proof reading it.

Thinkersbluff, 2020-12-24:
> Fixed it. Thanks for proof reading it.

Thank you for writing such complete and readable documentation.  I hope more people take the time to read it before posting issues that are already answered therein.

Sebazzz, 2020-12-24:
> 
> 
> > Fixed it. Thanks for proof reading it.
> 
> Thank you for writing such complete and readable documentation. I hope more people take the time to read it before posting issues that are already answered therein.

You and me both, mate 😉

Thinkersbluff, 2020-12-24:
“ Pre-heat PLA and ABS temperatures are properly shown in the touch screen interface”
ABS label was changed to PETG in the stock Creality version.

Thinkersbluff, 2020-12-24:
"Heaters are turned off when the actual probe is done to decrease electrical noise, which then causes Z-offset and bed leveling inaccuracies"

I find that phrasing to be ambiguous.  
If I may, I suggest this version may be more clear:

"Heaters are turned off while the actual probing is done. This reduces electrical noise which may otherwise cause inaccuracies in the Z-Offset and Bed Leveling measurements."

Sebazzz, 2020-12-24:
Thanks, updated it

Beeble2695, 2020-12-25:
Good write-up!
Only point i have resulting from the discord chat there was about terminology:

Support M701/M702 filament load/unload. With @grobux changing it to Extrude/Retract the terminology should be part of the sentence.

Thinkersbluff, 2020-12-25:
“ If you go to the home/info screen though, you can see it heats ups though.”
Typo. Should read, “ If you go to the home/info screen, though, you can see it heat up.”

Thinkersbluff, 2020-12-25:
“ After flashing go to the menu and reset to factory settings or issue M502 followed by M500 in Octoprint - do this also if you have been running an earlier release. You need to relevel the bed, reset the z-offset, and reprogram esteps (if you've changed them from defaults).”

I recommend you add “redo PID” to the list of things to be done after flashing, in step 8.  

I use M303 E0 S205 U and M303 E-1 S50 U routinely after M502. The U saves me having to enter the PID parameters, and could be coded into a small .gcode file to print from the SD card if people don’t use their serial port.

Thinkersbluff, 2020-12-25:
Since G28 now turns off the heaters while probing, and some folks have G28 in their start code, it might help folks to include some advice about reconfiguring their starting gcode to move M109 to after G28 in their start code, if they are currently using Cura and/or M109.  It may even be appreciated if you add a section wth recommended start and stop gcoding with explanations.

flofreud, 2020-12-25:
>  It may even be appreciated if you add a section wth recommended start and stop gcoding with explanations.

That. I assume that a recommendation of (commented) start and stop GCode will reduce the support requests :)

Sebazzz, 2020-12-25:
> 
> 
> > It may even be appreciated if you add a section wth recommended start and stop gcoding with explanations.
> 
> That. I assume that a recommendation of start and stop GCode will reduce the support requests :)

I'Il add a link [to this page](https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping) - but that G28 cools down after or overrides temperatures is a blocker IMHO, we can't ask people to reslice their stuff.

Thinkersbluff, 2020-12-25:
> I'Il add a link [to this page](https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping) 
That is a great resource page.  
Hopefully, an FAQ will reduce the probability that a disappointing number of users still post bug reports when they notice that their prints are starting before the nozzle temps get back up to their target settings.

> but that G28 cools down after or overrides temperatures is a blocker IMHO, we can't ask people to reslice their stuff.
Maybe I am different that way - I slice every time.  I don't maintain an inventory of past gcodes that I would not want to "reslice".

Sebazzz, 2020-12-26:
I've made some additional changes and added some imagery.

Botap, 2020-12-27:
Hi, I really would like to try the release 5 beta but for some reason I am missing the point on how to download the files. I can't see the 'assets' section :-( Hope you will be able to explain to me. Thanks in advance

Thinkersbluff, 2020-12-27:
> Hi, I really would like to try the release 5 beta but for some reason I am missing the point on how to download the files. I can't see the 'assets' section :-( Hope you will be able to explain to me. Thanks in advance

These a pre-release notes.  There are no assets yet, because it is not released yet.  If you would like a early look at CF5, you can find a developmental version on their Discord channel.

ThisIsMyUniqueUsername, 2020-12-28:
I tried this and it resolved several issues - Z-offset is now saved after reboot, but only when set from the main menu, not the value adjusted during print - could this be also be saved permanently? 

Also I used to set my z-offset to 0.15-0.2, now it's down to 0.1 for correct squish, is this the result of "improved homing accuracy"?

The second issue I noticed is the heater turned off during homing. Because it takes so long my nozzle temp can drop drastically before the print starts, especially in a cold room (it's winter here). Basically the nozzle preheats to 200'C, then homes and by the time it starts extruding the line on the right it's down to about 170'c. With large prints it has enough time to get back to the working temp on skirt, but printing a small item can be problematic. 

Does turning off the heater during homing really improve accuracy that much? I've never had issues with it before, so I'm not sure if the tradeoff is worth it.

Thinkersbluff, 2020-12-28:
With Flow Rate and Feed Rate both now having the same icons, and both being measured in %, it makes the footer display less clear.  Might want to consider a way to clarify that?

Sebazzz, 2020-12-28:
> 
> 
> With Flow Rate and Feed Rate both now having the same speedometer icons, and both being measured in %, it makes the footer display less clear. Might want to consider a way to clarify that?

This is your cue @grobux 😀

grobux, 2020-12-28:
Your comment is on point, I'm already thinking about a new design for the footer. There is much more infos to display than before, and the current design isn't suited.

grobux, 2020-12-29:
> 
> 
> With Flow Rate and Feed Rate both now having the same speedometer icons, and both being measured in %, it makes the footer display less clear. Might want to consider a way to clarify that?

I kept things pretty basic and replaced icons with labels, as I'm lacking space and creativity (for now at least).

The branch `extui-dev` of `CR6Community/CR-6-touchscreen` has the new design.

Sebazzz, 2020-12-29:
Feel free to go on icon hunt @Thinkersbluff 😉

grobux, 2020-12-29:
Is the fan speed missing/required on this screen?

Sebazzz, 2020-12-29:
Fan speed is indeed missing

Met vriendelijke groet,
Sebastiaan Dammann

Thinkersbluff, 2020-12-29:
> Feel free to go on icon hunt @Thinkersbluff 😉

Inspired by Fig 3 in this article: https://link.springer.com/article/10.1007/s40430-020-02610-0

[image]

[image]

I something like this is readable at the desired scale, it may also work for Flow Rate?

Sebazzz, 2020-12-30:
Let's release this baby.

closed 2020-12-30 by Sebazzz