CR6Community/Marlin #85: Selecting Load Filament before Unload Filament, in firmware-crealityv452-20201223-002158.bin, traps User on wrong screen.

Thinkersbluff, 2020-12-24:
New with the Pre-Release version of v5.
If the User selects Load Filament before selecting Unload Filament, the printer switches to an INFO screen once the nozzle achieves its pre-heat value.  That screen looks and behaves as though it is part of the Filament Change function. (Red button says "Load Filament".  Four long beeps + one longer beep.  Pressing the button produces no activity.  User is obliged to power off/on to recover control.

### Steps to Reproduce

1. Power-On printer
2. Select Prepare menu.
3. Select Filament Extruder menu.
4. Select value box & enter any positive value.
5. Press "Load"
6. Wait for nozzle temperature to reach target value.
7. Printer switches to INFO screen.

**Expected behavior:** 

At step 7, the printer should proceed with raising Z axis, Loading the entered mm of filament, etc..

**Actual behavior:** 

Printer switches to INFO screen and user has no recourse but to cycle power off/on to recover control.

#### Additional Information

See also https://discordapp.com/channels/759603746270609428/791082544899555358/791168120819679263

IF User uses the Unload FIlament function first, then Load Filament works as intended & this bug does not occur.

Sebazzz, 2020-12-24:
It should be resolved now. Not using M701 for loading filament but just a regular E axis movement, because no messages are being pumped from DWIN while the feed happens.

Thinkersbluff, 2020-12-24:
Yes, the original bug is resolved by your change to G1 from M701.
Is there any need to ensure E motions are relative, not absolute, when extruding this way? Or to Reset the extruder counter first?

Sebazzz, 2020-12-24:
> Is there any need to ensure E motions are relative, not absolute, when extruding this way? Or to Reset the extruder counter first?

Before issueing G1, I first set E to relative, then restore the setting afterwards ;)

Thinkersbluff, 2020-12-24:
> > Is there any need to ensure E motions are relative, not absolute, when extruding this way? Or to Reset the extruder counter first?
> 
> Before issuing G1, I first set E to relative, then restore the setting afterwards 😉
Ah great, thanks.  Did not see that in the Terminal readout.

closed 2020-12-24 by Thinkersbluff