CR6Community/Marlin #86: First autohome with latest btt source sits and waits 34~ seconds before lowering to bed

NickDevoctomy, 2020-12-24:
- *Be sure you have fully read the release notes and flashing instructions before posting an issue here*
- *Check if there is an existing _open_ issue that describes your problem and add your comments there*
- *Check if your issue has been resolved on the latest source code - or if there is a closed issue pointing to the next release*
- *Check that you have flashed the correct firmware for your device*

### Description

using branch extui-btt-skr-stock-tft
When running first autohome after powering on

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. power on printer
2. autohome

**Expected behavior:** 

1. nozzle raise a little
2. autohome x & y axis simultaneously
3. nozzle go to centre of bed
4. nozzle raise a little
5. nozzle lowers to bed to zero z axis

**Actual behavior:** 

1. nozzle raise a little
2. autohome x & y axis simultaneously
3. nozzle go to centre of bed
4. nozzle raise a little
5. 34~ second pause
6. nozzle lowers to bed to zero z axis

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files if you don't run a precompiled build
* Which hardware configuration do you use? CR-6 SE or MAX? Which motherboard? Stock TFT, BTT TFT?
* Provide pictures or links to videos that clearly demonstrate the issue.
* Which *exact* firmware version did you flash? Provide a link to the download you have used.

__Issues that do not follow this issue template will be closed__

NickDevoctomy, 2020-12-24:
Just to add, this seems to only happen from the touch screen.  So I will try updating that too to see what happens.

NickDevoctomy, 2020-12-24:
Actually, looks like I have the newest one already so please disregard my last comment.

NickDevoctomy, 2020-12-24:
It's now heading the bed and nozzle during autohoming too...

NickDevoctomy, 2020-12-24:
Aaah, that's the cause for the delay.  It's heating the nozzle and bed...  Shouldn't it be doing that during levelling, not autohoming?

Sebazzz, 2020-12-24:
No, it should always heat the nozzle to prevent plastic from interfering with the Z-offset. 

I think @grobux was working on getting the temperature display on the homing screen as well. That should clarify it - but we should probably add an additional text as well.

NickDevoctomy, 2020-12-24:
Okay, if that's now expected then great, it's just different from how it previously worked.  Thanks for clarifying.

closed 2020-12-24 by NickDevoctomy