CR6Community/Marlin #87: Auto bed levelling no longer working with stock touch screen UI, branch extui-btt-skr-stock-tft

NickDevoctomy, 2020-12-24:
- *Be sure you have fully read the release notes and flashing instructions before posting an issue here*
- *Check if there is an existing _open_ issue that describes your problem and add your comments there*
- *Check if your issue has been resolved on the latest source code - or if there is a closed issue pointing to the next release*
- *Check that you have flashed the correct firmware for your device*

### Description

Auto bed levelling would previously auto home, head bed and nozzle, and then level the bed.  During which the UI would show which point is being levelled on the bed.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Power on
2. Auto bed level

**Expected behavior:** 

1. Autohome
2. Heat nozzle and bed to 120 / 50 respectively
3. Test bed matrix and display on the UI as it progresses (at previously set temps)

**Actual behavior:** 

1. Autohome
2. Heat nozzle and bed to 120 / 50 respectively
3. Screen flashes up levelling matrix
4. Levelling matrix disappears
5. Nozzle and bed temps set to 0
6. Printer runs though bed levelling

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files if you don't run a precompiled build
* Which hardware configuration do you use? CR-6 SE or MAX? Which motherboard? Stock TFT, BTT TFT?
* Provide pictures or links to videos that clearly demonstrate the issue.
* Which *exact* firmware version did you flash? Provide a link to the download you have used.

I've attached a sample gcode of a calibration cube that worked as expected with my previous version.  Unfortunately I cannot revert to my previous version as I didn't keep a copy and can't locate it in the revision history, so I can't do any checks against the previous version I had.

[CCR6SE_xyzCalibration_cube.zip](https://github.com/CR6Community/Marlin/files/5739322/CCR6SE_xyzCalibration_cube.zip)

__Issues that do not follow this issue template will be closed__

NickDevoctomy, 2020-12-24:
I can confirm that printing still seems to be working as expected after the levelling hicup, although it completed **very** slowly. Much slower than the previous version.

Sebazzz, 2020-12-24:
Are you able to reproduce and collect some logging? I think @grobux was able to do it once but not reproduce it after.

> I've attached a sample gcode of a calibration cube that worked as expected with my previous version. Unfortunately I cannot revert to my previous version as I didn't keep a copy and can't locate it in the revision history, so I can't do any checks against the previous version I had.

How is the calibration cube relevant to this issue? 😉

NickDevoctomy, 2020-12-24:
The calibration cube is very relevant to the issue as it shows how I'm also running auto bed levelling within a print.

Including any preheat logic that I have going on.

NickDevoctomy, 2020-12-24:
I will have a go at generating some logs and provide them here as requested, will try and do that today.

Thinkersbluff, 2020-12-24:
I can confirm that the first Autobed Levelling after flashing the firmware also behaves this way on the stock CR-6.  Not just a BTT issue.  Subsequent leveling seems to work correctly, if it is not meant to stay on the 16-point display screen at the end of the cycle.

Sebazzz, 2020-12-24:
So it only happens after a factory reset?

Met vriendelijke groet,
Sebastiaan Dammann

NickDevoctomy, 2020-12-24:
I'll try running it a few times in a row to see what happens...

NickDevoctomy, 2020-12-24:
Just ran it twice in a row, first time showed the behaviour described above, and second time worked.  I must admit, I never recall this on the creality board firmware (community version) tbh.  Once it's done its second run, I'll reboot the machine and see if it works first time then.

Thinkersbluff, 2020-12-24:
> So it only happens after a factory reset? Met vriendelijke groet, Sebastiaan Dammann
> […](#)

On my stock CR-6, I flashed the v5 pre-release, then ran Autobed Level and it worked correctly.
I then did a factory reset from the menu and the system behaved as described in this bug.

Seems to be a function of whether or not there is an existing mesh when the routine starts?

NickDevoctomy, 2020-12-24:
Worked perfect after the reboot too, thanks.  That just leaves the speed, but then I guess that's probably personal preference so I'll play about with it a bit to suit my requirements.

Sebazzz, 2020-12-24:
Got the log. It appears that after factory reset we quickly get the all the meshes through from Marlins bed leveling code, which confuses the touch screen.

Sebazzz, 2020-12-24:
[image]

Sebazzz, 2020-12-24:
I think I got it now - please try the latest sources and retry.

@InsanityAutomation That call for `ExtUI::onMeshBedLeveling` start needs to be moved down. There is no other way for the touch screen to distinguish between Marlin resetting the bed leveling mesh before probing and the actual probing itself, agree? I've changed this in 5e2edd0 but I'm not sure how the guys upstream think about that change.

Thinkersbluff, 2020-12-26:
Looks like my system (stock CR6 TFT + 4.5.2 MB) is not yet performing as intended.  

G29 from terminal ends on wrong screen (see log) though does run ABL in synch with display, 
1. On first run of G29 after M502, M500, heaters are off and ends on PRINTING screen.  
2. On 2nd run of G29 after mesh 1, heaters are on. Launched while on PRINTING screen and Ends on PRINTING screen.
3. RUN ABL from LEVELING screen after capture of mesh 2 ends on LEVELING screen, heaters are on.
4. RUN ABL from LEVELING screen after capture of mesh 3 ends on LEVELING screen, but heaters are off on first ABL after Factory Reset.

[Issue87_25Decextui Test-log.txt](https://github.com/CR6Community/Marlin/files/5742701/Issue87_25Decextui.Test-log.txt)

Circumstantial evidence from small sample suggests heaters on or off changes the mesh values:
[MESHES_1-4.txt](https://github.com/CR6Community/Marlin/files/5742699/MESHES_1-4.txt)

NickDevoctomy, 2020-12-26:
Just got to try the latest fix and it didn't work as expected, this time it heated the hot end, but not the bed.  Then when it started it turned off the hot end heater.  Although it did continue to show the progress of the levelling process, so that bit is working now.  For now I'll just run it twice after a factory reset.

Sebazzz, 2020-12-26:
@NickDevoctomy Can you detail the exact steps to reproduce?

NickDevoctomy, 2020-12-26:
1. Flash firmware
2. Power on
3. Auto bed level

Same steps, nothing has changed, just being explicit here about this being the first time ABL is run from the screen.

Thinkersbluff, 2020-12-26:
“ ...the first time ABL is run from the screen.” 

No factory reset required to trigger this bug on your system?  Worked first time on my system but behaves in this bug if I then do a factory reset.

NickDevoctomy, 2020-12-26:
@Thinkersbluff sounds like you can reproduce the issue?

Thinkersbluff, 2020-12-26:
> @Thinkersbluff sounds like you can reproduce the issue?

Looks that way.  Factory reset seems to reliably trigger this bug on my system.

NickDevoctomy, 2020-12-26:
Cool, my printer's in the process of printing unfortunately, so I can't mess about with it right now. I can try a simple factory reset and ABL once that's complete.

Thinkersbluff, 2020-12-26:
[image]

Not quite level?  Tried printing using 4th mesh captured with above log. Some lines not quite fused. Some lines look quite “blobby”, where the slicer (Cura 4.8) went back to fill in the gaps it left.  Z offset left at 0.20, so that might be slightly off, too.  Generally happy though, as this will be a replacement cover for the back of the TFT so defects not visible.

Thinkersbluff, 2020-12-27:
> > @Thinkersbluff sounds like you can reproduce the issue?
> 
> Factory reset seems to reliably trigger this bug on my system.

UPDATE  Happy to report that when I flashed my display with the latest* pinned DWIN_SET files, I lost the ability to reproduce the bug.  (*The files had flashed were timestamped 22 Dec.  The new files I flashed are timestamped 23 Dec.)

First "Run ABL" after Factory Reset ran perfectly.
G29 from terminal ran perfectly.

I think this problem may be fixed.

Sebazzz, 2020-12-27:
@NickDevoctomy Can you please confirm this as well? 
[CR-6-touchscreen-2020-12-27.zip](https://github.com/CR6Community/Marlin/files/5744788/CR-6-touchscreen-2020-12-27.zip)

Thinkersbluff, 2020-12-28:
RATS - Using the display firmware CR-6-touchscreen-2020-12-27.zip + the latest motherboard F/W source as of 10pm Montreal time 27 Dec, the first ABL after Factory reset is again without heaters. :(

Sebazzz, 2020-12-28:
Got it! I was now able to reproduce that the heaters are turned off after probing. 

Exact reproduction steps:

1. Reset printer to factory settings
2. Reboot printer (M997)
3. Start print with G28,G29 in start gcode
4. After G29 completes, the heaters are turned off.

Sebazzz, 2020-12-28:
I thought it was the hot-end idle timeout but it isn't. 

Didn't capture anything useful either:
```
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X225.00 Y10.00 Z6.77
<<< Probe::probe_at_point  X225.00 Y10.00 Z6.77
Mesh level index: 16
Saving DWIN LCD setting from EEPROM
echo:Settings Stored (747 bytes; crc 65530)
//action:notification Settings Stored
SetNewScreen: 37
echo:busy: processing
//action:notification issue87-cube.gcode
  current_position= X225.00 Y10.00 Z6.77 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X225.00 Y10.00 Z6.77
>  X225.00 Y10.00 Z6.77
<<< do_blocking_move_to  X225.00 Y10.00 Z6.77
  current_position= X225.00 Y10.00 Z6.77 : > probing complete
Bilinear Leveling Grid:
      0      1      2      3
 0 +0.217 +0.134 +0.054 -0.026
 1 +0.161 +0.072 -0.042 -0.103
 2 +0.086 -0.010 -0.085 -0.153
 3 +0.006 -0.104 -0.150 -0.162
G29 uncorrected Z:6.77
 corrected Z:6.78
  current_position= X225.00 Y10.00 Z6.78 : sync_plan_position
<<< G29  X225.00 Y10.00 Z6.78
echo:G1 X0 Y0 Z15 F5000.0
echo:G1 Z2.0 F3000
echo:G1 X0.1 Y20 Z0.3 F5000.0
echo:G1 X0.1 Y200.0 Z0.3 F1500.0 E15
Writing to file: /PLR
```

Sebazzz, 2020-12-28:
This should now be fixed. I need to submit this bug upstream because this affect anyone with these settings.

closed 2020-12-28 by Sebazzz

Thinkersbluff, 2020-12-29:
Not sure if related, but I started RUN ABL with system heated to PETG pre heat values on the latest CF5 builds and the target temps changed to 120/50 at point 2.  Point 1 still displayed the PETG target temps.  

I know - don’t do that, then...  Just sayin’ in case ;)

Sebazzz, 2020-12-29:
> Not sure if related, but I started RUN ABL with system heated to PETG pre heat values on the latest CF5 builds and the target temps changed to 120/50 at point 2. Point 1 still displayed the PETG target temps.

I can't reproduce it. If you have a good scenario, please open a separate issue.

Thinkersbluff, 2020-12-29:
> I can't reproduce it. If you have a good scenario, please open a separate issue

Will do, if it persists.  Thanks.

Sebazzz, 2020-12-29:
@Thinkersbluff It appears to be an upstream bug:

When the heaters are paused for probing, they aren't actually paused, the `HOTEND_IDLE_TIMEOUT` function is used for expiring the timers. But now the idle temperatures have been set to 120C/50C it resets to that.

Thinkersbluff, 2020-12-29:
Gotcha.  That explains it.  Thanks.

Sebazzz, 2020-12-29:
Reported upstream:

- https://github.com/MarlinFirmware/Marlin/issues/20595
- https://github.com/MarlinFirmware/Marlin/issues/20596

Requested option to keep heaters of the bed on:

- https://github.com/MarlinFirmware/Marlin/issues/20598

Implemented already in this code base.

commit b1147ad5 (2020-12-29): Prevent hot-end idle timeout settings from interfering with PROBING_HEATERS_OFF

Previously the HOTEND_IDLE_TIMEOUT was (ab)used for disabling the heaters, but HOTEND_IDLE_TIMEOUT doesn't necessarily turn off the heaters. It just may set them to a lower temperature. In addition, some people like to level at a high temperature and we do not want to reset their temperature.

Fixes #87 also