CR6Community/Marlin #89: [FR] Option to exit yellow "Homing Failed" screen

ThisIsMyUniqueUsername, 2020-12-25:
### Description

When the probe fails to home properly (for example with some last-second spring sheet bed adjustment, or triggering the strain gauge by touching the nozzle to wipe oozing filament), the screen turns yellow and displays Homing Failed message, then the printer initializes cooldown cycle.

Power cycle is the only way to exit this screen, which causes more time waiting to preheat the bed and nozzle again.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Start a print or ABL
2. Cause the homing to fail by touching the strain gauge on first or second probe
3. Screen turns yellow and displays Homing Failed message.
4. Power cycle the printer and print or execute ABL again.

**Expected behavior:** 

The Homing Failed screen should have "OK" or similar button to go back to the main menu and try again.

**Actual behavior:** 

Homing Failed screen does not have an option to exit it.

#### Additional Information

firmware-20201106-192839.bin flashed on CR-6 SE with 4.5.2 board

[image]

Sebazzz, 2020-12-25:
This is possible but undesirable. When this screen is displayed Marlin has gone into the kill() state. This happens after homing, failed probing, on M112, or other conditions like thermal runaway. 

While we could take control at this point and endlessly pump messages from the touch screen and invoke a reset, we must tread carefully not to compromise the safety of the printer.

ThisIsMyUniqueUsername, 2020-12-25:
> This is possible but undesirable. When this screen is displayed Marlin has gone into the kill() state. This happens after homing, failed probing, on M112, or other conditions like thermal runaway. [...]

Understandable, thank you for the clarification.

Thinkersbluff, 2021-02-20:
> Understandable, thank you for the clarification.

@ThisIsMyUniqueUsername - Can this issue be closed, now, or would you still like something to be changed in the firmware?

Sebazzz, 2021-02-24:
Closing it, Feel free to re-open.

closed 2021-02-24 by Sebazzz