CR6Community/Marlin #90: G28,G29 in start gcode may cool down the nozzle if temperature is set before

Sebazzz, 2020-12-25:
Happens on unreleased firmware 5 beta.

Sebazzz, 2020-12-25:
@frenck I think you reported this

Thinkersbluff, 2020-12-25:
Is this not because of the probing now being with heaters off? I moved the M109 to after G28 in my start code, to make sure printing waits for the temp to recover.

Sebazzz, 2020-12-25:
You'd think so, but I've moved the heating code to before the G28 and it didn't turn off the heaters and just homed and started printing.

Thinkersbluff, 2020-12-25:
Well, I’ve been wrong before, so looking forward to learning the real story.  I still have the stock extruder fan and find it cools quite quickly when homing.  If Z=20 when homing starts, the nozzle temp drops several degrees further than if homing starts at Z=10, on my system.  Your performance may vary.

frenck, 2020-12-26:
So, what I (at this point) think that happened, is that I've preset a temperature (120/50) and the printer is heated/warmed up on that.

After that, started a print (Octoprint). It goes straight into doing the homeing sequence, including the Z, as temps are reached already.

Next, it immediately started printing, after homing and didn't wait for the actual printing temp to be reached.

---

So, I've been trying to re-produce this morning, in order to get some good re-production steps to share. Unfortunately, I've been unable to right now, nevertheless, it did happen to me twice.

It might as well have been a glitch in the matrix or a bug in the Christmas spirit. I'll keep an eye out in the upcoming days.

(I run a BTT board and BTT TFT)

Sebazzz, 2020-12-26:
Thanks for trying. Glad to hear (or maybe not) it doesn't happen always.

frenck, 2020-12-26:
I've been trying really hard, I've dug into the code as well.
If this is an issue, it should be an issue with Marlin itself, as it has protection for this (and it is enabled).

It should not be able to print when the nozzle temp is below 170, which in my case, it was.

Considering that, and my incapability to reproduce it at this point, I would opt to close this issue.

I'll open up a new one once it happens again, or reproduction was found.

Sorry to waste your time on this guys, nevertheless, thanks for approaching this issue 👍 

Happy Holidays 🎄

Sebazzz, 2020-12-26:
No problem. Best to chase these kind of bugs before release 5 is formally released to the general public. Thanks for testing and happy holidays! 😀

closed 2020-12-26 by Sebazzz