CR6Community/Marlin #93: [BUG] Z-offset in LEVEL menu skips some steps

ThisIsMyUniqueUsername, 2020-12-26:
### Description

Changing Z-offset option in LEVEL menu skips 0.15 and 0.19 (possibly more)

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Power on the printer
2. Tap LEVEL button
3. Tap UP and DOWN arrows under Z-offset

**Expected behavior:** 

I expected the Z-offset to change with 0.1 increment across the range

**Actual behavior:** 

Z-offset jumps from 0.20 to 0.18 and from 0.16 to 0.14, skipping 0.19 and 0.15 setting. See attached video file.

#### Additional Information

I'm running CR6Community-v2.0.1.7 firmware installed using firmware-20201106-192839.bin file and screen firmware installed using CR-6-touchscreen-2020-11-06.zip file on CR-6 SE with 4.5.2. board.

__Issues that do not follow this issue template will be closed__

https://user-images.githubusercontent.com/76540672/103159847-28e31700-47c6-11eb-8052-3c9032635672.mp4

Sebazzz, 2020-12-26:
This is resolved in CF5 after factory reset.

ThisIsMyUniqueUsername, 2020-12-26:
> This is resolved in CF5 after factory reset.

Thank you for prompt reply, do you mean factory reset from the printer screen menu? I was worried this option will wipe my custom firmware.

Sebazzz, 2020-12-26:
No, factory reset only restores the default settings baked in the particular firmware you're trying to flash 😉

Thinkersbluff, 2020-12-26:
> This is resolved in CF5 after factory reset.

Have you tried going past 0.28 to 0.30 on your latest firmware?  Mine still displays 0.29mm twice on the way up and down.

ThisIsMyUniqueUsername, 2020-12-26:
Sorry I just tried factory reset and it did not resolve the issue, so this bug report should remain open.

The factory reset option worked, I can tell by the Preheat PLA & ABS back to their default values from what I set and saved them previously.

Sebazzz, 2020-12-26:
I see it, I can reproduce it. Probably something odd with the way floats are represented in DWIN 😐

ThisIsMyUniqueUsername, 2020-12-26:
> I see it, I can reproduce it. Probably something odd with the way floats are represented in DWIN 😐

I'm glad it's not just me then. :) Could it possibly cause a random issue when a finished print job still generates a "Resume previous print DWIN technology" error after a power cycle? It happened to me twice but it's hard to reproduce, I'm not sure if I should report it as a separate bug in this case.

Sebazzz, 2020-12-26:
> It happened to me twice but it's hard to reproduce, I'm not sure if I should report it as a separate bug in this case.

You should report that upstream essentially. That means the PLR file was not removed by Marlin.

ThisIsMyUniqueUsername, 2020-12-26:
> upstream

I'll try to find a pattern to this behaviour and report it. So far it seems random, it happened both when I pressed "Finish" button at the end of the job and when I simply powered off the printer. Not a huge deal but worth noting.

Sebazzz, 2020-12-26:
@ThisIsMyUniqueUsername I think this is related to v4.5.2 motherboards that have trouble reading SD cards in general.
If you have a serial monitor attached you will also find "delete /PLR failed" emitted by Marlin.

ThisIsMyUniqueUsername, 2020-12-27:
> @ThisIsMyUniqueUsername I think this is related to v4.5.2 motherboards that have trouble reading SD cards in general.
> If you have a serial monitor attached you will also find "delete /PLR failed" emitted by Marlin.

Unfortunately somehow drivers provided by Creality don't work with my printer, it also reports incorrectly as "Unknown USB device" instead of "FT232R USB Adapter". I'm waiting for my Raspberry Pi to arrive to test it with Octoprint.

ThisIsMyUniqueUsername, 2020-12-27:
The latest pre-release still skips 0.17.. One thing fixed, another broken.

Sebazzz, 2020-12-27:
Fixed in latest sources.

closed 2020-12-27 by Sebazzz

Thinkersbluff, 2020-12-27:
> Fixed in latest sources.

Do you update the pinned .bin files in the Discord channel or are you refering to the GitHub extui branches, when you say “latest sources”?

Sebazzz, 2020-12-27:
I did say sources, didn't I 😉 I'Il post a new series of builds soon

Thinkersbluff, 2020-12-27:
Thank you.  I assumed you meant GitHub, but last time I seemed to have misunderstood. It looked like you had updated the Dwin_Set bin after I downloaded it from Discord, based on the file time stamps.   I have not learned how to compile that firmware.

Thinkersbluff, 2020-12-28:
> Fixed in latest sources.

I confirm printer no longer shows 29mm twice when adjusting Z Offset from 28 to 30 or 30 to 28, with the 27 Dec versions of MB and Display firmwares.

commit f0962a6c (2020-12-27): Z-offset is skipping one number - fixes #93

We need to round values before we send them to the display. Otherwise they get truncated which is perceived as one number being skipped.