CR6Community/Marlin #94: [BUG] Power cycle during ABL preheating causes the ABL to be stuck

ThisIsMyUniqueUsername, 2020-12-26:
### Description

When the printer is turned off and back on again during ABL preheat, starting the process again after reboot will cause it to freeze after preheating.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Start ABL and wait for the head to move to the centre and start preheating
2. Power cycle the printer
3. Start ABL again

**Expected behavior:** 

The ABL should be performed

**Actual behavior:** 

The head moves to the middle, bed & nozzle preheat, then head moves briefly to position 1 of the ABL process and goes back to the middle where it stays forever. I waited for several minutes after the preheating finished. See attached video.

The only way to end this cycle was to do factory reset from the screen.

https://user-images.githubusercontent.com/76540672/103160544-74012800-47ce-11eb-8d38-947865af6879.mp4

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files if you don't run a precompiled build
* Which hardware configuration do you use? CR-6 SE or MAX? Which motherboard? Stock TFT, BTT TFT?
* Provide pictures or links to videos that clearly demonstrate the issue.
* Which *exact* firmware version did you flash? Provide a link to the download you have used.

__Issues that do not follow this issue template will be closed__

Sebazzz, 2020-12-26:
Please follow the issue template and fill in the missing information. Thanks.

closed 2020-12-26 by Sebazzz

ThisIsMyUniqueUsername, 2020-12-26:
> Please follow the issue template and fill in the missing information. Thanks.

Done!

Sebazzz, 2020-12-27:
Okay it has probably failed to level. This should be fixed in CF5. In [the issues here](https://github.com/CR6Community/Marlin/milestone/2?closed=1) there should be a link for a firmware to try, otherwise head to the #prerelease-channel in the Discord. At the top you'Il find several files.

ThisIsMyUniqueUsername, 2020-12-27:
> Okay it has probably failed to level. This should be fixed in CF5. In [the issues here](https://github.com/CR6Community/Marlin/milestone/2?closed=1) there should be a link for a firmware to try, otherwise head to the #prerelease-channel in the Discord. At the top you'Il find several files.

Thank you, I will try that. I managed to reproduce the issue today - the nozzle homes fine in the middle (two touches, pressure sensor light comes on), then moves to 1st point of ABL and immediately back to the middle where it sits. 

Failed fixes:
- power cycle
- reset to factory

Working fix:
- power cycle then reset to factory

Now the ABL works.