CR6Community/Marlin #96: Filenames superimposed on home screen

ritchiedc, 2020-12-27:
### Description

When pressing "Print" home screen remains with filenames superimposed

### Steps to Reproduce

1. Fresh clone of CR6Community Marlin and CR-6-touchscreen on 27 Dec 2020
2. Build Marlin in VSCode and copy firmware in .pio\build to sd card
3. Build CR-6-touchscreen using build.cmd and copy contents of zip to micro sd card.
4. Turn off printer
5. Insert micro sd card with screen firmware, turn on printer and wait for "End!" (see screenshot).
6. Turn off printer
7. Remove micro sd card from screen and insert sd card with printer firmware
8. Turn on printer and wait for firmware to be written with home screen showing
9. Turn off printer
10. Remove sd card with fimware and insert sd card with gcode files.
11. Turn on printer
12. Press home screen print button.

**Expected behavior:** 

Display file list on

**Actual behavior:** 

File list is superimposed on home screen (see attached screenshot, sorry for the glare)

#### Additional Information

[image]
[image]

Sebazzz, 2020-12-27:
Your touch screen isn't flashed correctly. Note how the home screen doesn't contain the new designs of @grobux. Please retry flashing 2 more times.

closed 2020-12-27 by Sebazzz

grobux, 2020-12-27:
At step 3, make sure you are on `extui-dev` git branch of the repo `CR6Community/CR-6-touchscreen`.

ritchiedc, 2020-12-27:
Changing to the extui-dev branch worked. Thanks