CR6Community/Marlin #100: [BUG] New Filament Extruder "Load" (Extrude +) function does not raise Z before extruding

Thinkersbluff, 2020-12-28:
### Description

If Z =0.0 and select Extrude+, would prefer not to extrude until raise nozzle a bit.

### Steps to Reproduce

1. Move Z to0.0 (e.g. to check offset with paper)
2. Select Filament Extruder menu
3. Select Extrude+

**Expected behavior:** 

Expected printer to raise nozzle +10 mm

**Actual behavior:** 

Printer extruded at Z=0.0
:(

Sebazzz, 2020-12-28:
Well... Don't do that 😉

Sebazzz, 2020-12-28:
No going to fix this for CF5 beta unless I happen to have some time.

Thinkersbluff, 2020-12-29:
> Well... Don't do that 😉

Agreed.  My bad.

Sebazzz, 2020-12-29:
> Wouldn't be a problem, if M701 had worked correctly. Upstream bug?

Not necessarily. It was M701 before but how the code is currently structured, is that we don't pump messages while a handler (like the handler that runs when you press "Load filament") is running. But if that was working then you would probably negatively surprised by *beep beep beep* and a message stating you need to load filament. 

So from this point I see to options:
1. Restructure the code so we can run pump messages while a handler is running
2. Emulate M701 by raising the Z-axis too

Thinkersbluff, 2020-12-29:
> So from this point I see two options:
> 
> 1. Restructure the code so we can run pump messages while a handler is running
> 2. Emulate M701 by raising the Z-axis too

Not knowing which option is on the "right path" toward your ultimate objective, I won't risk a definitive answer.
I can say I would have been "ignorantly blissful" if Extrude had fully emulated M701.

If M701 had actually worked correctly, I could have been happy with option 1, too.

Sebazzz, 2020-12-29:
In the end I decided to go for the following approach:

- Use M701 again
- When on the feed screen immediate confirm everything Marlin sends us (the confirmation for loading filament) and don't beep

commit 4c37cfc4 (2020-12-29): Automatically handle M701 filament load without buzzer or confirmation when on the feed screen - fixes #100
closed 2020-12-29 by Sebazzz