CR6Community/Marlin #106: [FR] Make "Heaters off when probing" user-configurable

Sebazzz, 2020-12-30:
### Description

It would be preferred to have the "heaters off when probing" configurable. For me and many others it increases accuracy, yet @Ranney1 reported issues with this functionality. It would be nice if it could be configured, for instance from the "leveling" screen.

As simple on/off toggle would do.

ThisIsMyUniqueUsername, 2021-01-04:
I agree, I never had an issue with inaccurate probing on stock firmware so all it does for me is slows down the print when the nozzle needs to preheat again after probing.

Sebazzz, 2021-01-04:
Yes, that's why two firmware builds have been released.

Sebazzz, 2021-01-27:
Okay, this took the better part of the evening. 

Implemented in MarlinUI and DWIN touch screen, saved to EEPROM. Temporary gcode `C001` assigned for allowing this to set from serial monitor.

[image]
[image]
[image]

 I chose to call it "Improve leveling accuracy" so people pick the right choice by default - and so they must explicitly make a choice to take the risk to make it inaccurate, and when they do that and complain [we can send them this](https://knowyourmeme.com/photos/1016674-baton-roue) 😉

All jokes aside. This is how it is looking - and although I didn't use pink comic sans I imagine @grobux will still have a look at it.

thaukap, 2021-01-28:
For me it would be totally fine to leave the heater off but while heating. But please move the nozzle a lot more up, and, even better, also to another location.

Usually while probing some material will be placed on the print bed. Then, while heating, the nozzle continuously drips out some plastic. On the middle of the print bed. And there is not enough room between the nozzle or the print bed to clean both of them.

Thinkersbluff, 2021-01-28:
@thaukap If you usually have PLA residue “dots” on the bed after probing, are you setting the nozzle temperature higher than the default 145C?  
@Sebazzz would it make sense to park the head at X=0, Y=235 at the end of the ABL run, to facilitate cleaning in these cases?

Sebazzz, 2021-01-28:
> For me it would be totally fine to leave the heater off but while heating. But please move the nozzle a lot more up, and, even better, also to another location.

That has already been picked up in #143 

> Usually while probing some material will be placed on the print bed. Then, while heating, the nozzle continuously drips out some plastic. On the middle of the print bed. And there is not enough room between the nozzle or the print bed to clean both of them.

Don't set the temperatures too high when probing (leveling or homing). Can't fight physics 😉

ThisIsMyUniqueUsername, 2021-01-28:
You can add a 30-50mm retraction to end gcode in your slicer, after each print the filament will back a bit. The purge line and skirt are enough to prime the nozzle again and you avoid dripping and having to reheat the nozzle to swap the filament.

That, or just pull it out slightly during the preheat process.

ThisIsMyUniqueUsername, 2021-01-28:
You can add a 30-50mm retraction to end gcode in your slicer, after each print the filament will back a bit. The purge line and skirt are enough to prime the nozzle again and you avoid dripping and having to reheat the nozzle to swap the filament.

That, or just pull it out slightly during the preheat process.

This is better than moving the nozzle higher or away from the middle between the preheat and probing. When you probe, the heaters go off so if the nozzle is higher, it will take longer to reach the bed. The process is already taking painfully long time, sometimes I need to print a small 20-25minutes item and I spend half of it just waiting for things to preheat, probe and preheat again...

If you retract the filament before probing, only a tiny strand of filament might drop, plenty of time between the first and second probe to grab it with a pair of long tweezers which is what I do.

thaukap, 2021-01-29:
@ThisIsMyUniqueUsername  Sounds reasonable, and there are a few good ideas.
@Thinkersbluff Since when is 145°C default for PLA? On all of my PLA roles it reads: "Printing temperature 190-220°C" ;)

There is always at the beginning, cold start, coming some material out of the nozzle. Physics or not. Even at 200°C, PLA. And that is placed onto the print bed during probing. Also in the stock firmware. And while the nozzle sits directly above, waiting 20-30 seconds for the correct temperature: There is plenty of time to clean everything, but there is no room to do it.

Otherwise: I like the firmware a lot, thanks @Sebazzz and the team for the hard work!!! I'm fighting other nasty problems. Two times in the last few days the printer was in an unstable state when powered up without a SD card. One it it stops feeding material. Never saw this with the stock firmware. But that is off topic here and I need to create a "steps-to-reproduce" list.

Thinkersbluff, 2021-01-30:
> @Thinkersbluff Since when is 145°C default for PLA? On all of my PLA roles it reads: "Printing temperature 190-220°C"

I guess my comment was not clear, sorry. 
he "default" to which I referred is the nozzle temperature of 145C set by default in the firmware for the ABL probing. (Previously 120C in previous versions of CF and in the stock Creality firmware.)  It is also possible to ABL at higher temps, and some users consider that a better practice, but it will leave those blits on the bed if hot enough to print filament.  The goal of the 145 is to try to soften any filament blob that might be left on the nozzle, so that it does not interfere with the probe taring, without making it so hot that it promotes active oozing.

Sebazzz, 2021-02-02:
Exactly - to prevent oozing just us a different start gcode.

Sebazzz, 2021-02-03:
Just for fun I ran some bed levels...

**Mesh with heaters on**
```
Recv:       0      1      2      3
Recv:  0 -0.009 +0.009 +0.060 +0.171
Recv:  1 +0.003 -0.072 +0.011 +0.136
Recv:  2 +0.165 +0.027 +0.147 +0.295
Recv:  3 +0.499 +0.356 +0.405 +0.628
```

**Mesh with heaters off**
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 +0.139 +0.107 +0.115 +0.222
Recv:  1 +0.127 -0.032 +0.021 +0.135
Recv:  2 +0.226 +0.055 +0.128 +0.255
Recv:  3 +0.510 +0.312 +0.337 +0.544
```

Quite a significant difference! The point at (0,0) is almost a layer height different!

ThisIsMyUniqueUsername, 2021-02-03:
> Just for fun I ran some bed levels...
>
> **Mesh with heaters on**
>
> ```
> Recv:       0      1      2      3
> Recv:  0 -0.009 +0.009 +0.060 +0.171
> Recv:  1 +0.003 -0.072 +0.011 +0.136
> Recv:  2 +0.165 [...]

There might be a different reason for it. Run it few times with and without heater and compare - the results will vary each time. Now remove the bowden tube and do the same, you'll notice that each pass is much closer to the others.

I've been running it several times with and without the PTFE tube:

-0.131 | -0.106 | -0.227 | -0.182
-0.135 | 0.009 | -0.107 | -0.157
-0.073 | 0.105 | 0.009 | -0.106
-0.004 | 0.117 | 0.042 | 0.039

2nd pass:

-0.109 | -0.118 | -0.217 | -0.186
-0.137 | -0.001 | -0.121 | -0.17
-0.064 | 0.131 | -0.027 | -0.115
0.061 | 0.259 | 0.077 | 0.038

3rd pass:

-0.116 | -0.121 | -0.215 | -0.132
-0.147 | -0.019 | -0.176 | -0.177
-0.061 | 0.096 | -0.076 | -0.103
0.08 | 0.148 | 0.023 | 0.074

And now with the PTFE tube removed:

0.004 | 0.003 | -0.11 | -0.092
-0.1 | 0.051 | -0.051 | -0.081
-0.094 | 0.101 | 0.033 | -0.063
-0.06 | 0.077 | 0.033 | 0.018

2nd pass:

0.002 | 0 | -0.117 | -0.087
-0.101 | 0.043 | -0.056 | -0.084
-0.096 | 0.092 | 0.02 | -0.076
-0.04 | 0.089 | 0.038 | 0.022

The explanation - when the strain gauge sensitivity is set low enough to be accurate, it is affected by the PTFE tube pushing and pulling on it with different force in each mesh probe position.

Sebazzz, 2021-02-03:
> The explanation - when the strain gauge sensitivity is set low enough to be accurate, it is affected by the PTFE tube pushing and pulling on it with different force in each mesh probe position. 

No, the taring is supposed to compensate with that - and it does.

I got more or less the same results trying it multiple times.

ThisIsMyUniqueUsername, 2021-02-03:
> > The explanation - when the strain gauge sensitivity is set low enough to be accurate, it is affected by the PTFE tube pushing and pulling on it with different force in each mesh probe position.
> 
> No, the taring is supposed to compensate with that - and it does [in my experience].
> 
> I got more or less the same results trying it multiple times.

Perhaps I adjusted mine to be too sensitive, but I ran 5 or 6 passes with and without the tube and no other changes, the results are much more consistent without it. I've done it on glass bed, my PEI sheet seems to be a bit less consistent for some reason. 
 I suspect it's because the sticker I applied to the metal sheet is quite soft so it will give in a bit under pressure.

ThisIsMyUniqueUsername, 2021-02-03:
Not a big deal anyway, with the glass surface I've done it few weeks ago and never had to touch it since. :)

ztakis, 2021-02-03:
Anyone tested the actual real height with feeler gauge?

ThisIsMyUniqueUsername, 2021-02-03:
No, but I've done a different test. It's not very scientific but I have a flat electronic kitchen scale which is quite accurate (tested with known weights) and as I was adjusting the strain gauge sensitivity I let it home on the scale few times to see the max reading. I got it down to around 130g force on 1st touch and 75-80 on the second, anything lower is way too easily triggered even by the filament weight in the ptfe tube during homing.

Thinkersbluff, 2021-02-03:
> the taring is supposed to compensate with that - and it does [in my experience].

I sort of understand how taring could remove the effect of any tugging on the hot end being exerted by the tube, filament, harness, etc., if those forces were always the same.  I think it will still leave a random error term in the data, though, proportional to any difference in that upward force from point to point, as the tube "straightens out a bit" when the head moves from I=0, to I=1 to I=2 to I=3.  Even the length of the Bowden tube - and whether there is filament in or or not - may start to contribute differences between individual users' experiences, if any error atributable to the tube's leverage proves significant.

I know I am presently scratching my head as to why the bed level test squares look so good in only one or two spots on my bed but they are clearly printed too high or too low at other spots.  Maybe trying ABL with the tube removed will help me improve on that...  Easy enough to try and report back.

ThisIsMyUniqueUsername, 2021-02-03:
> I sort of understand how taring could remove the effect of any tugging on the hot end being exerted by the tube, filament, harness, etc., if those forces were always the same. I think it will still [...]

Sorry if it's a dumb question but when does the taring occur?" Is it on the first probe in the middle of the bed before the ABL starts, or the first touch for each point? 

Even then the tube will never move in the exact way on 1st and 2nd touch. Just touch it gently and you can see the blue LED indicating the bed touch flash.

ztakis, 2021-02-03:
> I sort of understand how taring could remove the effect of any tugging on the hot end being exerted by the tube, filament, harness, etc., if those forces were always the same. I think it will still [...]

The tube also twists (turns, rotates) when the head moves on the x axis, at least mine does.
Just mark a dot with a sharpie to check it. I did try with the tube removed but the biggest difference in my measurements was bed temperature, room temp vs well warmed-up.

Thinkersbluff, 2021-02-04:
> Sorry if it's a dumb question but when does the taring occur?"

Important question.  We all need to understand this.

This is what I believe:

Marlin is basically using a "dead-reckoning navigation system" to keep track of where the tip of the nozzle is.  
**"Taring" is the function by which Marlin defines a specific position to be coordinate 0.0 on an axis.** 

In the CF implementation, Marlin is configured to understand that the X an Y axis end stops are each "found" by stepping the applicable motor in the counter-clockwise direction until the associated end stop switch is activated.  It is also configured to "bounce" when it first triggers that switch and to then approach it again, more slowly, until it activates a second time.

Upon the first activation of the end stop switch, Marlin "Tares that axis, to define a coarse 0.0 reference from which it can safely count the number of steps corresponding to the specified bounce distance. (e.g. 5mm in the cases of X and Y.) 
Starting at that bounced position, Marlin again drives the motor counter-clockwise until the end stop switch is activated a second time.  Upon the second activation, Marlin "tares" that axis and uses that position as the definitive 0.0 coordinate for the origin on that axis.  All subsequent moves on that axis are performed by stepping the motor the applicable number of steps to move the required distance and direction.  Moving the axis to 0.0 returns the axis to the last tared position.

Marlin keeps track of the Z axis position in a very similar way to X and Y.  It uses probe activation as the Z end stop switch. It bounces 2mm instead of 5mm, but otherwise the Z axis taring logic is the same as for X and Y.

When Marlin is performing ABL, it is also performing "taring", but in that case it is not defining the position of z=0.0 for the axis, rather it is measuring the local variations in Z=0.0 at the various probe points.  
At each ABL probe point, Marlin will:
1. sense the bed once
2. tare the local mesh point value,
3. drive Z upwards from that point by the distance specified, 
4. drive down to detect the bed a second time and subtract that position from the Axis z=0.0 position
5. tare the local mesh point value
6. drive Z upwards from that point by the distance specified, 
7. drive down to detect the bed a second time and subtract that position from the Axis z=0.0 position
8. Calculate the average of those two measurements, and set that average as the value of the local probe point's local deviation from the Z axis = 0.0 position on the lead screws.
9. Tare the probe a third time, to provide a reference value for raising Z before moving to the next probe point.
10. Move up to the defined position and move to the next probe point. 

Probably a longer answer than expected. Hopefully just long enough to remove any false assumptions, or to clarify any relevant details.

NOTE: all of the above is only the best of my knowledge.  If anyone reading this finds factual errors or omissions, please contribute qualified corrections below.

Sebazzz, 2021-02-04:
No this is incorrect. Taring refers to telling the strain gauge to take a new zero position.

Met vriendelijke groet,
Sebastiaan Dammann

Thinkersbluff, 2021-02-04:
Thanks for the clarification, but I could use a bit more help, then, to properly describe the sequence of events.
How would you rewrite what I wrote, with this correction incorporated?  I think we all need to understand exactly what the machine is doing when it is probing for Home, for Offset and for ABL mesh positions.

thaukap, 2021-02-04:
Interesting informations. thanks for sharing the info.

Does anybody know how "bending of the bed" is corrected? I mean, on my machine, especially on the outside of the bed, the nozzle bends the bed noticeable down when it touches it for probing. For sure a 1mm or more. But not in the middle of the bed. The sensor needs to force to get activated. And the bed mount is a little bit floppy on the outside. 

Are there some "compensation values" for this?

Thinkersbluff, 2021-02-04:
> Does anybody know how "bending of the bed" is corrected? I mean, on my machine, especially on the outside of the bed, the nozzle bends the bed noticeable down when it touches it for probing. For [...]

Better to ask this question in the Discord.  Issues should remain focused on the issue in the title.

Sebazzz, 2021-02-04:
@thaukap Bending of the bed is not corrected because the printer/firmware doesn't _know_ the bed is bending. Think of it from the perspective of the printer - it only has the limited sensors it has, so how _could_ it know the bed is bending?

That said - you need to adjust the sensitivity of your strain gauge if the bed is bending. It is probably set _way_ too high. Check [the chapter here on strain gauge calibration](https://damsteen.nl/blog/2021/01/19/mounting-the-biqu-h2-to-creality-cr6#strain-gauge-calibration) on how to do it.

Also, of course, because this leveling system is so susceptible to mechanical issues, make sure you've ruled that out (eccentric nuts etc).

@Thinkersbluff Rewriting what you wrote:

-----

Let me first define some things:

- __Optical sensor / opto sensor:__ On the gantry - normally lit red (except on BTT boards). Blocked by a metal piece when the gantry moves down.
- __Probe / strain gauge / Z-endstop:__ The metal piece [on which the hot-end is suspended](https://damsteen.nl/blog/2021/01/19/mounting-the-biqu-h2-to-creality-cr6#the-cr-6-se-bed-leveling-system-recap).

> Marlin is basically using a "dead-reckoning navigation system" to keep track of where the tip of the nozzle is.
__"Homing"__ ~~Taring~~ is the function by which Marlin defines a specific position to be coordinate 0.0 on an axis.

> In the CF implementation, Marlin is configured to understand that the X an Y axis end stops are each "found" by stepping the applicable motor in the counter-clockwise direction until the associated [...]

Yes, that second bump is safety and accuracy measure:
1. If the motor is moved too far beyond the initial probing point, it means either the endstop is broken or triggering too early.
2. The second bump is done slower. As we know that mechanical switches and probes in general are not _very_ accurate at higher speeds, the second bump is allowing more accuracy.

> __For X/Y:__ Upon the first activation of the end stop switch, Marlin ~~tares~~ _sets an initial zero position for_ that axis, to define a coarse 0.0 reference from which it can safely count the number of steps corresponding to the specified bounce distance. (e.g. 5mm in the cases of X and Y.)

> Starting at that bounced position, Marlin again drives the motor counter-clockwise until the end stop switch is activated a second time. Upon the second activation, Marlin ~~"tares" that axis and~~ [...]

You are forgetting the homing offsets here. In the default firmware, the endstops are actually at -5 and -2 for X and Y respectively. So, endstop activation for X means X=-5. That makes that endstop + 5mm is X=0.

> Marlin keeps track of the Z axis position in a very similar way to X and Y. It uses probe activation as the ~~Z end stop switch~~ _the optical sensor_. ~~It bounces 2mm instead of 5mm, but otherwise the Z axis taring logic is the same as for X and Y.~~ 

No, this is incorrect. 

First:
1. The optical sensor is often incorrectly called the endstop. It is not. On the CR-30, which uses the same sensor it is indeed the endstop. Not on this printer - the "fixed" probe is the endstop. 
2. The optical sensor fulfills a non-critical function in the probing for home (note: non-critical I'm mean - functionally you can leave it out in the firmware and homing would work fine). However, if the firmware listens to this sensor, it is critical that it works correctly. The sensor acts as a cue for the firmware to perform two actions - when it is blocked by the gantry:

  - The probe/strain gauge is tared (like taring or zero-ing your kitchen scale). Remember that the strain gauge in the kitchen scale and in your CR-6 is not much different.
  - From that point onwards the Z-endstop is enabled - before the sensor is blocked the Z-endstop (=probe) is left out of consideration. Now you also know why either a optical sensor or probe fail can cause the probe to ram into the bed: When the optical sensor fails to let the firmware know it is being blocked, the firmware never bothers to listen to the Z-endstop (=probe).

With this knowledge it is also easy to see:

- Why a failed strain gauge board that "failed closed" (= always sends the endstop signal) causes the homing position to lift immediately when the optical sensor is blocked.
- Why moving the optical sensor up or down is complete rubbish.

Also good to note the the _point of the strain gauge triggering while it bends_ plus the correction of the _Z-offset_ determines the _Z=0 position_. 

> When Marlin is performing ABL, it is also performing "taring", but in that case it is not defining the position of z=0.0 for the axis, rather it is measuring the local variations in Z=0.0 at the various probe points.
>At each ABL probe point, Marlin will:

>   _Tare the probe_
>    Sense the bed once
>    Remember the local mesh point value
>    Drive Z upwards from that point by the distance specified,
>   _Tare the probe_
>    Drive down to detect the bed a second time ~~and subtract that position from the Axis z=0.0 position~~
>    _Take a weighted average (2 / 3) of the two positions_

Note that this is dependent on the probing configuration. This procedure describes how the community firmware and mostly stock is configured.

>   ~~ tare the local mesh point value~~ Set that average as the mesh point value relative to the homing position.

With this in mind you can also see why that having a negative or positive only mesh is incorrect.

>    Move up to the defined position and move to the next probe point.

Thinkersbluff, 2021-02-04:
Thank you for making time to correct my write up.  
This will help me keep things straight in my head and hopefully be useful for others in future.

Cheers.

thaukap, 2021-02-07:
@Sebazzz: You made my day, thanks a lot. Calibrated the strain gauge as described. Now the bed didn't tilt anymore when it's probed on the edges. I took the skirt from my last print and it differs by 0.03 in height, end to end, 7 cm or so. Not 100% but perfect for me. Used the original glass plate from Creality.

Further, using your script from [here](https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping), solved the other issue with spots of material right in the middle while heating up the nozzle. Also great! After a material change let it cool down and clean the nozzle during that process. Before the next print is started.

---
Had the strange behaviour that the Z-axis when tuning to values below zero during print jumped back to zero all the time. But I will wait to check again and report this until it happens again in the next version. AFAIR there was a bug that the printer crashed when changing the value too fast. Maybe that is related.

Thanks again!

Sebazzz, 2021-02-14:
Closing this as this has been implemented.

closed 2021-02-14 by Sebazzz