CR6Community/Marlin #109: Stuck on loading screen after firmware update

willrc627, 2020-12-30:
### Description

After updating the touchscreen and mainboard firmware from 20201106-1928-39.bin and CR-6-touchscreen-2020-11-06, the printer is stuck on the new boot loading screen (looks slick by the way) and won't progress any further.

### Steps to Reproduce

1. Update touchscreen firmware from 2020-11-06 to 2020-12-30 using steps provided:
1a. Power off printer
1b. Format micro SD card as FAT32 w/ 4096 sector size in Windows
1c. Place DWIN_SET folder in root of SD card
1d. Safely remove SD card from Windows and install card in touchscreen
1e. Turn on printer and wait for "DONE!" to appear as in "flash_successful.jpg" [NOTE: File counts were 1,3,3,0,1,**_5_**,0,0,0,0]
1f. Power off printer, remove SD card
2. Update mainboard firmware from 2020-11-06 to CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19 using steps provided:
2a. Format regular SD card as FAT32 w/ 4096 sector size in Windows
2b. Place "firmware-20201230-141905.bin" in root of SD card
2c. Safely remove SD card from Windows and install card in mainboard
2d. Power on printer
3. Stuck on new boot loading screen with 0% progress showing on the bar
4. Wait 5 minutes to be certain that any firmware changes would've completed
5. Turn off printer, remove SD card, turn on printer,
6. Stuck on new boot loading screen with 0% progress showing on the bar

**Expected behavior:** 

Expected to see progress on the bar and see the new firmware in action.

**Actual behavior:** 

Stuck on new boot loading screen with 0% progress showing on the bar

**Logging:**

Logging pending

### Additional Information

Harware: Stock CR-6 SE w/ v4.5.2 mainboard, no mods.
Firmware: CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19 [Linked here](https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip)

[image]

Sebazzz, 2020-12-30:
Try using a different SD card, renaming the firmware bin file (important!). Reformatting the SD card.

willrc627, 2020-12-30:
Could it be possible that the 2020-11-06 firmware on the mainboard hasn't actually read the BIN file from the SD card? The firmware .BIN file on the SD card looks untouched and no new "EEPROM.DAT" file is created? Not sure how I would verify that or force an update if not. I've also tried multiple SD cards, will try renaming the bin file now. Thanks!

grobux, 2020-12-30:
I got a similar situation in the past. If it persists after following @Sebazzz advice, you can try to downgrade your motherboard's firmware to the official 1.0.2, and then flash again the community version.

Sebazzz, 2020-12-30:
The bootloader reads the SD card. Marlin doesn't. The Creality stock main boards give no indication of firmware flashing.

Try flashing the old firmware and new one again. Also make sure you don't have any other (hidden?) bin files on your card.

willrc627, 2020-12-30:
> Try using a different SD card, renaming the firmware bin file (important!). Reformatting the SD card.

Quickest resolution I've every gotten. I appreciate everything you've done for the community @Sebazzz! Renaming the bin file from "firmware-20201230-141905.bin" to "firmware.bin" worked and now I'm up and running!

Can this step be included in the official release notes?

closed 2020-12-30 by Sebazzz