CR6Community/Marlin #110: Part fan control issue

unixluv, 2020-12-30:
### Description

Part fan setting does not accept fan speeds from Cura. Fan speed must be manually overridden from front panel after the print has started.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Turn on fan from Cura, and save in gcode.
2. Print file on CR6SE
3. Note that fan speed is 0%
4. Manually set fan speed to what you set in Cura.

**Expected behavior:** 

Fan speed should be able to be set independently for each model.  If a fan setting is not set by gcode, then it should take the printer's default value

**Actual behavior:** 

Fan speed setting in gcode appears to be ignored.

**Logging:**

I don't use serial or octoprint due to issues with USB ports.

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

I have a 4.5.2 stock mainboard. I have a all metal extruder and capricorn tubing.  Nothing else different from stock.

I am using community firmware CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19/firmware-20201230-141905.bin  I also flashed the touchscreen from community CR6-touchscreen-2020-12-30.

Sebazzz, 2020-12-30:
Can you create a minimally viable gcode file to reproduce? I haven't seen this issue. The fan is off at the first layer, then slowly ramps up at subsequent layers.

grobux, 2020-12-30:
To complete @Sebazzz answer, can you also provide your test file as a Cura project, so we can check everything.

unixluv, 2020-12-31:
It could be user error with the new version.  Let's close it for now and let me play around with it.

closed 2020-12-31 by unixluv