CR6Community/Marlin #111: X-axis, Y-axis and Z-axis values are displayed in cm on Print Screen, but labelled as being in mm

Thinkersbluff, 2020-12-30:
###Description

The new Print Screen at CF5 displays the target positions of X, Y and Z in "real" time, but the values appear to be "off" by an order of magnitude.

Suspect the displayed values are actually reported in mm but being displayed in cm with a mm label.

### Steps to Reproduce

1. Print from Host
2. When printing screen appears, X-axis, Y-axis and Z-axis positions are being reported in cm, but are labelled mm

**Expected behavior:** 

All axis positions should be displayed in mm.

**Actual behavior:** 

X-axis, Y-axis and Z-axis positions are reported in cm, but labelled as mm.

### Additional Information

Using stock CR-6 SE + stock Creality TFT, flashed with 30 Dec 2020 release of CF5 firmware on MB and on display.

https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

Photo will be uploaded, momentarily

Sebazzz, 2020-12-30:
Ha! @grobux is ahead of you!

Sebazzz, 2020-12-30:
Letting @grobux close this if it is fixed.

Thinkersbluff, 2020-12-30:
> Ha! @grobux is ahead of you!

Provided for the record only.  Sorry to be a day late.

grobux, 2020-12-30:
Thanks for you report @Thinkersbluff, testing a fix right now.

grobux, 2020-12-30:
Fixed by [CR-6-touchscreen@9d50e16](https://github.com/CR6Community/CR-6-touchscreen/commit/9d50e1608eb63fccd26a7e7dcd7c7a30010c97d4)

closed 2020-12-31 by Sebazzz