CR6Community/Marlin #114: Print Screen does not display Z-offset

ztakis, 2021-01-01:
### Description

Print Screen does not display Z-offset

**Expected behavior:** 

All other info is shown, why not Z-offset?

Sebazzz, 2021-01-01:
This is by design. In our opinion the Z offset is not necessary compared to the other data points. You set and forget, or tune it from the tuning screen.

ztakis, 2021-01-01:
I believe it is helpful as a reminder, especially if someone is using various nozzles from different manufacturers.
And more than once I have change it in real time when I noticed that the first layer didn't came out as expected.

Sebazzz, 2021-01-02:
Changing it in real time would still happen in the tuning screen.

Assigning this to @grobux to see if we still have some space somewhere for this.

ztakis, 2021-01-08:
May I suggest something like this:

[image]

dryphi, 2021-01-08:
> I believe it is helpful as a reminder, especially if someone is using various nozzles from different manufacturers.
> And more than once I have change it in real time when I noticed that the first layer didn't came out as expected.

How do you change the Z-height in real time?

ztakis, 2021-01-08:
By using the Tune screen.

Thinkersbluff, 2021-01-10:
Since we can now tap to change many of the parameters displayed in the footer of the Printing Screen, the only two parameters we need the Tune screen for are LED ON/OFF and Z Offset...  I could live with shorter pushbuttons if it created the missing real-estate for those controls on the Printing screen.

ztakis, 2021-01-10:
That's a great idea! Why not abolish the Tune button all together, then make the Pause and Stop buttons horizontal, one next to the other like the CONFIRM/CANCEL buttons. Then there should be enough space for LED ON/OFF and Z Offset.

Thinkersbluff, 2021-01-10:
To play “Devil’s Advocate”, though, the current ability “to tap to modify” those parameters is not self-evident, while the “Tune” button is...

Sebazzz, 2021-01-10:
I agree. The tap to modify function is just a little extra for the ones who know how to find it. And the tuning screen is still there for firmware retraction, LED control etc.

Thinkersbluff, 2021-01-10:
This seems like a relevant test-case for structuring decisions about the future menu tree, under issue #121

ztakis, 2021-02-24:
In order to close this, I submit these variations of the printing screen which I think have the potential to satisfy most users and seem not too difficult to implement:

[image]

[image]

[image]

Sebazzz, 2021-02-24:
Good input, thank you.

I like the second the most, but I do worry about the ability of DWIN to render the X/Y/Z values that small _and_ readable. Needs some experimentation I guess. For the status I'd abbreviate Linear Advance Factory to simply LA. 

More personal opinion: I don't like the LED button in the middle of the other values; I also doubt the value of needing to have that LED so accessible. Regardless, it looks out of place to me 🙂

ztakis, 2021-02-24:
For Linear Advance Factor you might use K-factor also but LA is fine too.
As for the LED, you might want to replace it with the retraction override button.

rockymtnpatriot, 2021-02-24:
@ztakis I like that location of the LED button as well as the rest in your last screenshot.

Sebazzz, 2021-02-24:
> For Linear Advance Factor you might use K-factor also but LA is fine too.

I thought of that, but the K-factor is an even more obscure term. K refers to the K in the linear advance formula. 

For the retract override we currently have these icons: 

[image]
[image]

Sebazzz, 2021-02-24:
BTW: I would opt for a combination of the older design and the second design: X/Y/Z at the bottom, LED in the top bar, and a retract override indicator at the status part.

ztakis, 2021-02-24:
I agree for that combination.
For retract override you have these toggles too

[image]
[image]

 We can make something squarish out of that

Sebazzz, 2021-02-24:
Yes, perhaps for the status part just retain the icon without the label.

ztakis, 2021-02-24:
[image]

Sebazzz, 2021-02-24:
Yes, exactly like that 😄

ztakis, 2021-02-24:
Glad you like it :)

Sebazzz, 2021-02-24:
So, going forward: 

I'Il assume you don't want to get into that DWIN rabbit hole, and also assume you mocked that up in Photoshop/GIMP 😉

Would it be possible to update the [appropriate PSD here](https://github.com/CR6Community/CR-6-touchscreen/tree/extui/gfx) with the screen (without the M117 and X/Y/Z placeholders). and create a "print paused" and "print finished" variant of this as well? For import in DWIN BMPs need to be exported from the PSD as well. (FYI @grobux )

Thanks again.

ztakis, 2021-02-24:
Believe it or not I made all this with Paint 3D. But yeah I'll give it a try with gimp.

ztakis, 2021-04-13:
@Sebazzz Shouldn't we close this?

Sebazzz, 2021-04-13:
You're right. don't know how this slipped through the net 😄

closed 2021-04-13 by Sebazzz