CR6Community/Marlin #115: Print Screen shows feed rate not printing speed

ztakis, 2021-01-01:
### Description

Print Screen shows feed rate not printing speed

**Expected behavior:** 

Print Screen shows feed rate instead of printing speed which is confusing as Tune Screen shows the expected printing speed adjustment. I think it would be better that Print Screen to display just FLOW  SPEED  FAN.

Sebazzz, 2021-01-02:
This is about the inconsistency in the labels I guess? The underlying data is the same: Marlin has no concept of print speed - only feed rate.

Thinkersbluff, 2021-02-01:
When we added control over Flow Rate, we also introduced the Feed Rate label, without going back & making other screens consistent with that term.  

As an Engineer, I find units of "%" make more sense for Rates than for Speeds.
Users with no prior experience of CNC or RepRap jargon are probably less comfortable with arcane terms like "Feed Rate", but it is certainly embedded in Marlin and in their LCD menus.  What terminology are you using for the BTT TFT Marlin mode menus?

To me, it makes a lot of sense to retain consistency with the legacy terms  Standardizing on RepRap vocabulary should facilitate a more rapid understanding of all 3D printing resources for newcomers.

Sebazzz, 2021-02-05:
Yes - so update the Tune screen by replacing "Printing speed" with "Feed rate". 

@grobux I think you are quicker than me. Come to think of it, the bed temperature and nozzle temperatures use the same icon - is that intentional?

grobux, 2021-02-05:
No problem, I'll do it.

> The bed temperature and nozzle temperatures use the same icon - is that intentional?

A bit of laziness maybe. 😋 Also, in the icon lib used, the main differences around the "temperature" keyword is basically the height of the temp gauge. Anyway, I'll check that too.

grobux, 2021-02-05:
Fixed by https://github.com/CR6Community/CR-6-touchscreen/commit/d8d9133131b85af8eb88ae01f196fad8cd1f2bc2

I took the opportunity to make new custom icons for bed and nozzle temperatures. Very simple, based on the ones in the footer and a bit larger than other menu icons, but more clear I guess. I put them everywhere they were needed. I'm open to better ideas and suggestions.

[image]

jwrayburn, 2021-02-06:
> Fixed by CR6Community/CR-6-touchscreen@d8d9133

Fantastic! Thanks!
Jerry

ztakis, 2021-02-06:
May I suggest some icons for flow rate?

[image]

Sebazzz, 2021-02-06:
Flow rate is extrusion rate, so I would suggest something that looks like an extruder. Marlin has already a similar icon for generic TFTs, [check here - second picture](https://damsteen.nl/blog/2020/11/06/flash-marlin-on-anet-et-printers#booting).

Sebazzz, 2021-02-14:
@grobux Are these icons useful to you? I think the gray part is like a transparency key.

[image]
*Flow rate [extrusion multiplier]*

[image]
*Feed rate [print speed]*

[image]
*Hot-end*

[image]
*Bed*

[image]
*Bed (heating)*

More icons can be found in the [Marlin/src/lcd/tft/bitmaps folder](https://github.com/CR6Community/Marlin/tree/extui/Marlin/src/lcd/tft/bitmaps).

So to complete this issue:

- Rename "print speed" to "feed rate" (done)
- Perhaps take a look at the icons but that is up to you @grobux 🙂

Sebazzz, 2021-03-01:
@ztakis did some updates to the printing screen which I have committed to the repository. DWIN part is still to be done.

Sebazzz, 2021-03-03:
[image]

Current screen implemented by @ztakis with a bit of #208 mixed into it already.

For this item this has now been implemented and I'Il close this issue.

If anyone feels strong about feed rate vs print speed, flow rate vs extrusion multiplier, let's open a new issue for that.

Thanks again for @ztakis for implementing this.

closed 2021-03-03 by Sebazzz