CR6Community/Marlin #116: Make preheat timeout and pause while printing heater timeout user configurable

unixluv, 2021-01-01:
### Description

Various heater timeouts need to be user-configurable.  When preheating, sometimes the bed and nozzle need a good heat soak before printing with some finicky filaments. Currently, the printer will time-out and the preheat will need to be restarted.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Select preheat {PLA/PETG} from the Prepare menu.
2. Wait some time, like go get a coffee
3. Return to printer and discover that printer exited preheat and now shows 120C.

**Expected behavior:** 

During pause during printing and during preheat (maybe other conditions?) that the heaters will not timeout, or do so under a timeout value under user control.

**Actual behavior:** 

Printer heaters timeout and user has to restart the process.

**Logging:**

N/A

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

4.5.2 MB with CF5 firmware.  No crazy mods from stock except the creality all metal filament feeder.

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

N/A

Sebazzz, 2021-01-02:
For now I'd opt to increase the timeout to 10m.

Sebazzz, 2021-01-27:
I've increased the heater timeout to 30 minutes. This should be plenty and give a good balance between safety and between turning off too soon and interrupting people workflows.

closed 2021-01-27 by Sebazzz