CR6Community/Marlin #117: Bed Mesh levels are not stored or recovered succesfully after power cycle

Ranney1, 2021-01-01:
__Issues that do not follow this issue template will be closed__

- *Be sure you have fully read the release notes and flashing instructions before posting an issue here*
- *Check if there is an existing _open_ issue that describes your problem and add your comments there*
- *Check if your issue has been resolved on the latest source code - or if there is a closed issue pointing to the next release*
- *Check that you have flashed the correct firmware for your device*

### Description

Bed level (mesh) data is not used after power cycle.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Flash firmware
2. M502 > M500 (clear EEPROM, restore firmware settings)
3. G28
4. G29
5. Results of mesh data are visible. 
6. M420 V > Mesh data OK
7. Power off, power on
8. Send: M420 V
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling OFF

**Expected behavior:** 

Mesh data is restored after an power cycle if data is saved with M500.

**Actual behavior:** 

It looks like data is not stored of recovered succesfully.

**Logging:**

After G29

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 +0.042 +0.046 +0.043 +0.037
Recv:  1 +0.097 +0.041 -0.019 -0.081
Recv:  2 +0.134 +0.049 -0.049 -0.146
Recv:  3 +0.196 +0.075 -0.038 -0.178

Send: M500
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (746 bytes; crc 56933)
Recv: //action:notification Settings Stored
Recv: ok

Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 +0.042 +0.046 +0.043 +0.037
Recv:  1 +0.097 +0.041 -0.019 -0.081
Recv:  2 +0.134 +0.049 -0.049 -0.146
Recv:  3 +0.196 +0.075 -0.038 -0.178
Recv: 

POWER CYCLE

Send: M420 V
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling OFF
### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

BTT Mainboard with STOCK TFT

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

I have the same issue with an github version of the firmware without any changes. (latest version, config data for BTT mainboard and stock tft)

Ranney1, 2021-01-01:
Update:
Installed latest release (from prerelease-testing' channel: CF5beta-prerelease3-btt-skr-cr6-with-stock-creality-tft-2020-12-29-16-18 .zip (screen and mainboard firmware) ==> Still not working.

It looks like that this is only happening on a BTT board.

Ranney1, 2021-01-01:
Latest update,

After an power cycle, it is NOT working. But after issuing an M501 it is working. 

Example:
G28
G29
M420 V => OK
M500 => OK
M420 V => OK
POWER CYCLE
M420 V => Invalid mesh  => M501 => M420 V => All data is ok and restored

Ranney1, 2021-01-01:
Issue closed.. Sorry for this. I have found the problem in an octoprint plugin... Really stupid and did not realised it...

closed 2021-01-01 by Ranney1

Sebazzz, 2021-01-01:
Can you elaborate on that?

Ranney1, 2021-01-01:
Yes, of course. I was trouble shooting and found out that the mesh level is working after ABL, that I can activate it with an M420 V.
After some time I found that it was also working if I send an M501 first. So my idea was; it has to work because it is saved Succesfully and can be load successfully. My trigger was that marlin when connecting was reporting 'emty' mesh points. 
After a while, I found that octoprint was sending an M503 when connecting.(......) I've changed these settings and now everything is working fine (of course). Thanks for help and sorry that it takes your time!

OMG, how stupid can it/I be... 🙈