CR6Community/Marlin #118: Z position Display reads .02 instead of .2

philloaring, 2021-01-02:
It seems the Z axis position is off by one place. Layer height reads .02 when it should read .2__Issues that do not follow this issue template will be closed__

- *Be sure you have fully read the release notes and flashing instructions before posting an issue here*
- *Check if there is an existing _open_ issue that describes your problem and add your comments there*
- *Check if your issue has been resolved on the latest source code - or if there is a closed issue pointing to the next release*
- *Check that you h

### Description

*Description of the bug or requested feature*

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** 

[What you expect to happen]

**Actual behavior:** 

[What actually happens]

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Infoave flashed the correct firmware for your device*rmation

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Thinkersbluff, 2021-01-02:
Is this a duplicate of issue #111 ?  

If so, it will be fixed in the next release.
If not, please elaborate, per the temp;ate.

philloaring, 2021-01-02:
Yes it is a duplicate. Sorry, didn't see it. Thanks for the response!

Thinkersbluff, 2021-01-02:
No problem.  Please close the issue, @philloaring.

@Sebazzz - if we are going to close issues raised on the current release, it may help reduce duplications if you also add them to the "Known Issues" section of the Release Notes.

closed 2021-01-02 by Sebazzz