CR6Community/Marlin #119: Screen standby on default touch screen not activating

a-disease, 2021-01-03:
The standby feature once the machine has been turn off and back on again even during print screen stays completely lit. To reactivate the feature the standby feature must be toggled off then back on for it to work. 

Cr6se - 4.5.2 main board with default touchscreen. Using beta release 5 - https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip
Printer managed through octoprint.

Sebazzz, 2021-01-03:
Same underlying cause as #113 - and fixed in 83d8e7e.

For now either:

   - Go to PLA/PETG preheat settings and then click the back button.
   - Use gcode M501 (either through a file, or terminal, start gcode etc.) to make the settings in effect

closed 2021-01-03 by Sebazzz

Sebazzz, 2021-01-09:
In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try the bugfix ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)).