CR6Community/Marlin #120: Allow disabling S-Curve in menu

ilgrank, 2021-01-03:
Hello
Not really a bug, but, as explained in this post:
https://www.reddit.com/r/CR6/comments/kbjwh5/here_are_6_of_my_9_cr6s_in_my_business_print_farm/
it seems that Linear Advance and S-Curve do not work very well together
https://www.reddit.com/r/CR6/comments/kbjwh5/here_are_6_of_my_9_cr6s_in_my_business_print_farm/gfi6pvw?utm_source=share&utm_medium=web2x&context=3

Sebazzz, 2021-01-03:
Please continue in #83

Sebazzz, 2021-01-03:
S_CURVE_ACCELERATION is fairly complex to make just configurable, so I suggest to take that suggestion to Marlin directly. Other than that, please discuss the default movement kinematics settings in #83 - with evidence preferably 😉

closed 2021-01-03 by Sebazzz