CR6Community/Marlin #122: Add Marlin+Display firmware compatibility checks

Allan-N, 2021-01-03:
### Description

One common issue on the various forums point back to incompatible versions of the Marlin and Display firmware (e.g. the display update did not work, the Marlin update did not work, the wrong version was installed, etc).  It would be nice if we could detect these mismatches and report them to the user.

### Steps to Reproduce

1. Make no changes to the display firmware 
2. Insert SD card with new [community] firmware
3. Power on printer

**Expected behavior:** 

I "expect" things to just work but because I have not yet updated the display firmware that just isn't going to happen. But, I'm clueless that I also needed to update the display (or I didn't format the microSD card correctly and the update didn't work).  Here, it would be nice to tell me that the display firmware is not compatible with the current printer firmware.

**Actual behavior:** 

Odd behavior, missing functionality, blank screen, etc

### Additional Information

CR-6 SE
Creality 4.5.2 main board
Creality display

### Random Thoughts

From .../Marlin/src/lcd/extui/lib/dgus_creality/creality_touch/DGUSDisplayDef.h, we have VP_UI_VERSION_MAJOR, VP_UI_VERSION_MINOR, VP_UI_VERSION_PATCH, and VP_UI_FLAVOUR.  

Starting with the VP_UI_FLAVOUR variable, I would suggest we set a value of 0xCFCF ("community firmware").  Other versions/vendors (e.g. Creality, Tiny Machines, etc) would use other values.

For the other VP_UI_VERSION_xxxxx variables I would also suggest that we utilize (and increment) values as changes are made to the display firmware.

Now, with the display version numbers available we can have the Marlin firmware check to see if the display firmware is compatible.  If not compatible, we should switch to a page displaying a message.  One possibility would be to use the confirm page (DGUSLCD_SCREEN_CONFIRM) and use the message lines (VP_MSGSTR1, VP_MSGSTR2, VP_MSGSTR3) to convey an appropriate firmware-out-of-sync warning.

Bonus points if we didn't move the VP_UI_VERSION_xxxxx/VP_MSGSTR1/VP_MSGSTR2/VP_MSGSTR3 variable locations or the 66_Confirmation screen from that used in the Creality base (i.e. Community Marlin w/Creality Display firmware == visible warning!!!).

Ranney1, 2021-01-03:
Maybe that it can be an easy step to implement an dgus firmware version what is showed on the info screen together with the marlin firmware version?

Sebazzz, 2021-01-03:
For this we need a "kill screen" with a variable text, preferably shared between Creality display FW and previous versions of the community firmware. Thát is the challenge.

Even then, it is no guarantee of a correctly flashed display unfortunately.

Allan-N, 2021-01-03:
@Ranney1 Showing the DGUS firmware version would be good to know

@Sebazzz My thoughts also started with the "kill screen" but then I switched to using a "confirm" page (yes, there's an incompatibility but "OK, proceed" because maybe what's installed is close enough).

Allan-N, 2021-01-29:
Reading more and more about folks having flashing issues and, again, I'm thinking that it would be good to let folks know of the mismatch.

I read that the BTT board provides a visual indication when new firmware is being flashed (or something like that).  In a similar way, maybe we could also have Marlin flash the LED if the display firmware is not known to be compatible enough with the board firmware?  Could even use different flashing patterns to indicate marlin-running-newer-version, display-running-newer-version, etc.

Thinkersbluff, 2021-02-01:
> Maybe that it can be an easy step to implement an dgus firmware version what is showed on the info screen together with the marlin firmware version?

This sounds like a good first step, at least.  
IMHO, this team may not be able to fix whatever it is about motherboards that sometimes causes them to ignore the flashing process, but it is probably in our best interests to help users recognize whether the MB flashing process has succeeded.

At least the DGUS flashing process gives feedback that can be interpreted and compared with photos.  It may still not be 100% reliable, but it is more feedback than the MB flashing process offers, today.
I suspect that some users only "think" they have flashed the MB to a new release, because they see the progress bar on boot up after trying to flash it.   

If the MB is actually ignoring their flashing attempts, then the user is going to have a lot of frustrating and perplexing trouble as their display fails to "play nice" with the MB until/unless they eventually reflash the display back to whatever version is fully compatible with what remains on the MB.

I imagine the INFO screen could be easily programmed to display a unique version ID for each new release of the Display Firmware.
INFO already displays an MB version ID, which could easily be made unique.
If pressing the INFO button also transmitted the display version ID to the MB firmware, then the MB firmware could probably return a message that showed "Compatible firmware versions" or "Firmwares don't match", as appropriate.  Users could be advised to flash the display firmware first, as part of the process.

Implementing that feature would give users a helpful cue when their boards don't flash and it would enable Devs to ask simple questions of users who subsequently report difficulties flashing the latest CF release and would allow for future implementations to incorporate compatibility testing (e.g. by look-up tables) - or not - as you choose.

It probably breaks an inviolable rule to suggest this, but you might even be willing/able to recompile previous CF releases if you had to, to retroactively encode a unique firmware identifier into each, since users are still downloading and installing "old" CFs.

Sebazzz, 2021-02-03:
We discussed this at length in #prerelease-testing. Some relevant excerpts:

> 1. The motherboard (usually v4.5.2 boards) ignore the firmware and users don't have enough attention to notice that there was no delay.
> 2. The touch screen was (not) flashed correctly.
> 
> In both cases we have a sub cases, specifically:
> a. Not flashed at all
> b. It can be flashed but just partially - or corrupted.

> (b) corruption is impossible to detect.
> So let's say we only handle (a) - then we need to determine how to show that. I can imagine that users don't like their printer to blink every boot.

> Motherboard firmware not flashing can be easily "fixed": Show the version number in the M117 message on the main menu.
> DGUS not flashing at all: LED and beep, because we can't output through the display.
> That would then be "CR6Comm-Rel6-Pre3 ready"

Thinkersbluff, 2021-02-03:
FWIW, I also posted a question on the Marlin Discord as to whether it is practical to request a Marlin feature that somehow checks for corruption of the installation, like CRC.  I see there is an EEPROM CRC check routine, for instance, because people are citing "CRC Check Failed" troubles with it.

Sebazzz, 2021-02-03:
Implemented version checks for display and show firmware version in M117 on boot.

closed 2021-02-03 by Sebazzz