CR6Community/Marlin #123: Rapid Z Offset adjustment Cause Factory Reset

JFI611, 2021-01-03:
__Issues that do not follow this issue template will be closed__

- *Be sure you have fully read the release notes and flashing instructions before posting an issue here*
- *Check if there is an existing _open_ issue that describes your problem and add your comments there*
- *Check if your issue has been resolved on the latest source code - or if there is a closed issue pointing to the next release*
- *Check that you have flashed the correct firmware for your device*

### Description

When rapidly adjusting the Z offset, the printer will factory reset and a M112 code will display on the touchscreen.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Go to Z offset adjustment screen.
2. Rapidly adjust the Z offset. Either Up or Down. Doesn't Matter. This will also happen it adjusting durning a print.

**Expected behavior:** 

[What you expect to happen]

I expected the Z offset to adjust without issues.

[What actually happens]

**Logging:**

Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 11352)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 38001)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 32496)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 31811)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 14836)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 57071)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 65052)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 6476)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 58648)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 471)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 14091)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 59942)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: N0 M110 N0*125
Recv: PowerUp
Recv: Marlin CR6Comm-CR6-Rel5
Recv: 
Recv: echo: Last Updated: 2020-12-30 | Author: (Sebastiaan Dammann, CR-6 SE with stock motherboard)
Recv: echo:Compiled: Dec 30 2020
Recv: echo: Free Memory: 49079  PlannerBufferBytes: 1472
Recv: //action:notification CR-6 Community Ready.
Recv: echo:EEPROM version mismatch (EEPROM=ERR Marlin=V82)
Recv: SetNewScreen: 28
Recv: SetNewScreen (saving): 28
Recv: Mesh level index: 0
Recv: Mesh level index: 1
Recv: Mesh level index: 2
Recv: Mesh level index: 3
Recv: Mesh level index: 4
Recv: Mesh level index: 5
Recv: Mesh level index: 6
Recv: Mesh level index: 7
Recv: Mesh level index: 8
Recv: Mesh level index: 9
Recv: Mesh level index: 10
Recv: Mesh level index: 11
Recv: Mesh level index: 12
Recv: Mesh level index: 13
Recv: Mesh level index: 14
Recv: Mesh level index: 15
Recv: echo:Hardcoded Default Settings Loaded
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (747 bytes; crc 30424)
Recv: //action:notification Settings Stored
Recv: echo:EEPROM Initialized
Recv: Taring probe
Recv: //action:prompt_end
Recv: BL24CXX Check succeeded!
Recv: echo:SD card ok
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: M155 S2
Recv: ok
Send: M113 S2
Recv: ok

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

v4.5.2

CF5Beta-cr6-se-v4

Sebazzz, 2021-01-04:
Confirmed. For now - don't go crazy on the Z-offset 😀

closed 2021-01-04 by Sebazzz

Sebazzz, 2021-01-04:
This is fixed now :)

JFI611, 2021-01-05:
Will this just be part of the next release?

Sebazzz, 2021-01-17:
There is a prerelease in the #prerelease-testing channel in discord.

commit ea4337a9 (2021-01-04): Prevent rapid adjustment of settings causing a factory reset - fixes #123

Apparently doing rapid Z-offset adjustments and subsequently saving settings is _SLOW_. So slow, that Marlin is not able to tell the watchdog it is still alive. So the watchdog loses patience and resets the CPU. This also corrupts the EEPROM in the process.

To fix it, now only save the EEPROM in a 250ms loop when requested.