CR6Community/Marlin #125: Can't manually retract or extrude

JamesSF69, 2021-01-05:
Can't manually retract or extrude  when just threading filament

The stock CR6-SE filament extruder is tough to get the filament to pass through. SO I used to use the manual extract/push/extrude to unload and load the filament in/out.
When I try all it does is raise the z-axis.
It's not until the filament is fully loaded in with the extrude work.

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Push filament into eh extruder housing
2. Press the manual retract or extrude
3. Nothing happens

**Expected behavior is that the extruder motor will pull or push the filament.

**Actual behavior: sometimes it just raises the z axis.

V4.5.2 motherboard, community New years already compiled firmware both marlin and display.

Thinkersbluff, 2021-01-05:
Hi, @JamesSF69 JamesSF69.  Are you having trouble feeding the filament from the filament runout sensor to the extruder, or having trouble feeding filament all the way through the extruder?  (or both?)

The devs on this firmware team are very quick to translate feedback into corrections, so if there is a bug in the code they will find and fix it.  The part where it raises the  Z axis before extruding or retracting is intentional though, as designed by the Marlin coders to ensure that the extruder is not too close to the bed before extruding (even the retract process includes a brief extrusion to ensure that the filament is not stuck to the inside of the nozzle before retracting quickly.)

Just in case these tips are helpful:

I found this little thing completely solved my problems feeding from runout sensor into the extruder: https://www.thingiverse.com/thing:4594094

To get the filament to feed all the way through the extruder, I find it definitely helps to:
-  cut the filament on an angle (so there is no blunt end to catch on the edge of the filament clasp or extruder assembly at the exit hole
-  bend to remove any slight curve from the end of the filament (so it is less likely to get caught on the edge of the exit hole)
-  turn the filament a little and try again, if it gets hooked-up inside the extruder on the way out.

JamesSF69, 2021-01-05:
Thank you, but yes I know that the stock extruder is finicky. But with the normal firmware, I could push it in, tap the manual feed, and the extruder would pull the filament through, and then I would release the extruder and push the filament all the way in. And when removing filament it was nice to just tap the "retract" (once it was heated up) and have the software pull the filament out some of the ways to make it easier to get the filament out.
None of this works in the Community firmware for me. The only thing I can get it to do is when I tap either feed or retract is the z-axis moves up slightly. 
And this is odd as it obviously is feeding and retracting when I'm printing. Is there a sensor at the nozzle??

Thinkersbluff, 2021-01-05:
> Is there a sensor at the nozzle??

No nozzle sensor.  
How much did you ask it to extrude? (Should default to 100mm.)

I am having Load Filament issues with the silky PLA  filament I am printing, right now, so I specifically experimented with that function after flashing and it did work for me, so I am thinking there may be something else at play for you.  E.g. Any risk that the upward Z axis movement may also be pulling the filament out of the extruder teeth, before it tries to extrude?

BTW - Did you Restore Factory Settings after flashing?

MericasMeuth, 2021-01-06:
Not sure if I should open this as a separate issue, but when I manually enter something like 5mm onto the touchscreen and press retract, sometimes it will instantly retract, other times there is a long delay. In either case it is followed by a much longer extrude, which sends

https://user-images.githubusercontent.com/72102612/103715380-7877ce80-4f8e-11eb-9282-1d536f3fda56.mp4

 filament shooting out of the nozzle. The same scenario, except pushing extrude, doesn't do that.

JamesSF69, 2021-01-06:
Update.

I had (or so I thought) hit the "factory reset" when I first flashed the firmware. But decided to try again. This time I kinda got a second reaction. 

I'm still having odd issues, but it seems once the nozzle comes up to temp the "retract and extrude seem to work now, kinda.

BUT: It still causes the z-axis to rise and or fall. Extrude will send it down (which is annoying as it makes it hard to reach in and remove the wasted filament.

Also, it's still a bit intermittent. Even with the temp showing full heat, it sometimes still won't retract. Also, it will retract a bit then double the amount and extrude all in the same action.

And, when I enter a new "amount" i.e. 25mm, it does not take and reverts to 100. I have to cycle the machine off/on then enter the new amount for it to stick.

JamesSF69, 2021-01-06:
> Not sure if I should open this as a separate issue, but when I manually enter something like 5mm onto the touchscreen and press retract, sometimes it will instantly retract, other times there is a long delay. In either case it is followed by a much longer extrude, which sends
> 
>  Retraction.Error.mp4 
> filament shooting out of the nozzle. The same scenario, except pushing extrude, doesn't do that.

Yes, I touched on this as well.

MericasMeuth, 2021-01-06:
I don't have Octoprint running on anything, but I am currently installing it so I can pull get a log and post it here.

Sebazzz, 2021-01-06:
That it extrudes too much is fixed in the latest sources. But the Z-axis raise and lowering is expected - we're just using the standard Marlin filament load/unload gcodes.

> Update: Nevermind, installing Octoprint on windows is above my head.

You can just use Pronterface (included in PrusaSlicer).

JamesSF69, 2021-01-06:
> That it extrudes too much is fixed in the latest sources. But the Z-axis raise and lowering is expected - we're just using the standard Marlin filament load/unload gcodes.
> 
But the "raise lower(ing) is odd. As mentioned, when extruding, it lowers and almost touches the bed, and the extruded filament bunches up and could potentially stick to the nozzle and cause a mess.
It's not so much that it extrudes too much that it retracts then pulls the filament back in (quite a bit). Also, it doesn't remember the "amount" and always defaults to 100.

Sebazzz, 2021-01-06:
You'Il have to raise lowering the Z with the Marlin devs. I find that stupid too, but the alternative is not raising the Z-axis and then you have an issue too. Just don't load filament with the Z being so low. The stock firmware does not facilitate raising the Z at all, for instance.

> It's not so much that it extrudes too much that it retracts then pulls the filament back in (quite a bit). 

Can you explain? The extrude function extruded exactly 50mm too much. The retract function does extrude a bit and then retracts.

> Also, it doesn't remember the "amount" and always defaults to 100.

Should it remember anything? Stock doesn't either.

JamesSF69, 2021-01-06:
> You'Il have to raise lowering the Z with the Marlin devs. I find that stupid too, but the alternative is not raising the Z-axis and then you have an issue too. Just don't load filament with the Z [...]

JamesSF69, 2021-01-06:
I guess I never noticed that the stock firmware didn't remember the setting, hmm Must be my dementia kicking in as I really thought it did  LOL
Anyway, the rasing and lowering.
Yes I "raised" the Z manually when I changed my filament to avoid buildup and accessibility to the extrude.
This raises and lowers at will, sometimes raises, sometimes lowers. I can deal with that I guess, no biggie, I'll over-compensate the z height prior.
BUT, we still have the issue of extruding too much (The image shows a 25mm "extrude"), and the oddity that when "retracting" via the screen, it retracts then pushes the filament back in, almost double the amount of extraction.

[image]

Thinkersbluff, 2021-01-06:
Hi, James - You may be trying to use the Extrude/Retract commands to perform a task they are not currently designed to support.

If you have not found it yet, this Marlin page provides some details on how each of the commands work:
https://marlinfw.org/meta/gcode/

The Community FIrmware devs have designed the Filament Extruder control screen to allow the User to enter a length value (defaults to 100mm) and to either issue an M701 command (to load) or an M702 command (to unload) that length of filament.

I "get" that when you are trying to calibrate your extruder esteps, you need a function that just moves the filament precisely the distance you entered, nothing else.
The M701 and M702 commands, on the other hand, are designed to support Loading or Unloading filament into a printer, with certain specific use case design scenarios in mind. (and calibrating esteps is not one of those design scenarios)

e.g. To retract a filament from a cold nozzle, Marlin has designed M702 to overcome a couple of practical real-world issues, like:
- the filament will likely have a wider "knob" at the very tip, where it was pushed into the nozzle.  That knob may be too large to pull back through the Bowden tube.  If they extrude some filament first, and then pull it back, the retraction is more likely to succeed.

Because programmers like to code once and use often, Marlin developers also designed their Pause and Change Filament functionality and their Filament Load/Unload functionality to leverage one another.   Their M600 code can be used to interupt a print in progress, park the head away from the print, and prompt the user to change filament.  In the Configuration_adv.h file, provided in the configs folder of the firmware zip file, you can search for "#define ADVANCED_PAUSE_FEATURE" to review the parameters which M701 and M702 use by default, including as implemented by these Community Firmware devs for Filament Extrude/Retract.  

This part probably explains the "strange" behaviour you are seeing, when you tap the "Retract" button:
  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_PURGE_RETRACT       13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_PURGE_DELAY       5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.

If you just want a way to move the filament in & out via the extruder, then Retract/Extrude is not the function to use.  The way it works now is by design.  It is not a bug, it is a feature.

**I suggest you open a new issue, call it a Feature Request {FR} and ask the devs to add Extruder to the Move menu.  That option does exist on other printers and may be something they can agree to and can easily accomodate.**

Cheers

Sebazzz, 2021-01-06:
> **I suggest you open a new issue, call it a Feature Request {FR} and ask the devs to add Extruder to the Move menu. That option does exist on other printers and may be something they can agree to and can easily accomodate.**
> 

That won't be necessary - [in issue #57](https://github.com/CR6Community/Marlin/issues/57#issuecomment-753632049) a fully assisted e-step calibration is built 😉

keram79, 2021-01-06:
Wouldn't it be better then if the current Extrude / Retract would be called Load/ Unload to really reflect what is done, according to the Marlin GCodes?

JamesSF69, 2021-01-06:
I am not trying to do anything fancy. I am NOT using "codes" I'm just pressing the screen functions that are there, and they are doing odd things. this is not a feature request, this is a bug. Or at least what I see from other comments as well as an issue.  It randomly raises or lowers the z height. Sometimes it goes up, sometimes it goes down. It's not consistent. Along with when the screen button is pushed to "extrude" and I have it set to 0.25, or anything to that matter it extrudes about 300 mm of filament.
Also when I push the screen button to "retract" it will retract the correct amount, but then grab the filament again and pull it back in.

keram79, 2021-01-06:
My comment was directed to @Sebazzz and @Thinkersbluff , not you, @JamesSF69  :)
I guess when the current Extract and Retract that you're trying to use would correctly be named "load" and "unload" [Filament], you (and me!) would not have fallen for this trap.

MericasMeuth, 2021-01-06:
It's in the video I posted. In that case, I set it to retract 5mm. It retracted what seemed like 5mm and then shot out that long string of Filament.

JamesSF69, 2021-01-06:
Ok so I agree with Keram. Not I was directing that to @Sebazzz , I think now It's named incorrectly and confusing. It was fine for just moving filament before without all this "purging/retract/extrude. It was just a retract this amount, stop. Extrude this amount, stop.

keram79, 2021-01-06:
Basically, the Extruder mechanism of the CR6 doesn't need a Load / Unload functionality at all, as unlocking the extruder allows to push the filament in completely until it oozes out of the heated nozzle, or pull it back for removal likewise.
Having a real extract / retract function would instead be beneficial, e.g. for esteps calibration, or to stop oozing filament by retracting 5mm via Extruder instead of manually.

Thinkersbluff, 2021-01-06:
> Wouldn't it be better then if the current Extrude / Retract would be called Load/ Unload to really reflect what is done, according to the Marlin GCodes?

That was discussed at length in the Discord channel, before this design was chosen.  Fwiw, I voted for Load/Unload, to be consistent with the Marlin LCD menu design and the function names.  Extrude/Retract was preferred by others.  Can't please us all.

Sebazzz, 2021-01-06:
With #57 in place we should now revisit it anyway.

MericasMeuth, 2021-01-07:
Update: I was actually running 4 alpha with the 5 beta screen firmware. After correcting this, my extrude/retract work correctly.

Sebazzz, 2021-01-09:
In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try the bugfix ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)). The Z-axis will still raise and lower, like a native Marlin filament unload and load does, but the numbers should be more accurate.

Thinkersbluff, 2021-01-10:
I tried the new esteps function with the “free air” method described in Luke’s hot end fix document.  I have his fix installed. I use Capricorn tubing, an all-metal extruder cold end with the stock extruder gear and an otherwise stock CR-6 SE hot end and  0.4mm stock nozzle.
I removed the Bowden tube from the cold end of the extruder and measured the filament extruded from or retracted into the cold end.  

The extrude test calculated 98.7 steps if I extruded into the Bowden tube, but 95.1 if I extruded into free air. I think some advice may be necessary on how this test is meant to be performed, to help Users avoid inadvertently creating themselves a problem with friction in the system.

I also tried with and without the filament runout sensor in series with the extruder, just to be sure it did not impact the extruder’s performance. I saw no impact on the esteps.. 

Overall, I liked the interface.  If possible, it would help to give feedback like “E-Heating” while the nozzle is heating.  I was confused at first, when it said “Extruding” but it wasn’t.

I did notice that the esteps function extrude rate is much slower than is programmed for the load/unload function. I thought it much too slow. Is there a design reason for that rate?

I also tried the updated Extrude/Retract function. 
 The Retracted amount seems to be “short” by the amount extruded by the M702 command. I measured 92 mm retracted, not the 100 commanded.  That may confuse folks..
The Extruded amount was spot-on in free air”.  I could only extrude about 20mm through the hot end at that rate. Far too high a pressure in the nozzle.  Baffled as to why, if it is just me, but definitely a problem

closed 2021-01-26 by Sebazzz

Sebazzz, 2021-01-26:
I've added a status message in 4d73c0f1a5c6bc3f03195e26e7b02717e0a7530d. 

With this I consider the issue resolved (also given that we have an new e-step calibration now.

Keep watching the #prerelease-testing channel on the discord if you want to try it out. A new build will soon be published there.

commit 1cee5ee7 (2021-01-26): Manual retract or extrude: show progress on display

Fixes #125

Add P parameter to specify purge length for filament change command M701