CR6Community/Marlin #126: Octoprint can't set temperature

Grimmvg, 2021-01-05:
### Description

Extruder temperature can not be regulated or set via octoprint. After clicking the "accept V" in octoprint , it will show 0°C and the printer extruder temp will also stay at 0°C

### Steps to Reproduce

1) Start octoprint
2) Load file to print 
3) Press "Print" - the temperature stays at 0°C 
4) Press + / - or insert a value manually - temp jumps back to 0°C 

**Expected behavior:** 

Temperature to change to my setting on printer via octoprint commands

**Actual behavior:** 

temperature stays at 0

**Logging:**

(wasn't super sure how to do this)
Send: M111 S247
Recv: echo:M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
Recv: ok
Recv:  T:117.86 /0.00 B:54.49 /0.00 @:0 B@:0
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: echo:N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 M140 S60*82
Recv: echo:N1 M140 S60*82
Recv: ok

### Additional Information
Changed the start G code in Cura 4.8 from E29 to M420 S1 , [as suggested by](https://github.com/CR6Community/Marlin/issues/77) Sebazzz but did not solve the issue. 

CR6-SE - 4.5.2 motherboard (august '20) - Community firmware 5 beta
" CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip                    CR-6 SE (v4.5.2 motherboard)" 

## While trying to reproduce this after a first test print where the issue persisted , it seems like after the first print the issue is solved and temperature is controllable with octopi. Also changing the file then after is possible and the extruder temp is adjustable with octoprint and it does send the commands to the printer to heat up. Bit confused now what i did to "solve" it...

Sebazzz, 2021-01-05:
Do you have a minimal reproducible gcode file I can load?

Grimmvg, 2021-01-05:
[LEDHolder.zip](https://github.com/CR6Community/Marlin/files/5770277/LEDHolder.zip)
thats the one i was just using.

Not sure if it's related or a seperate issue , when i boot the printer, flow rate is at 0%. Also after i start a print it stays at 0% , and i don't see any difference in quality when i put it (manually) to 100% 

// Im kinda confused myself here, octoprint UI stayed on , just had a power cut , restarted the printer (no auto resuming anymore :( ) and now the temp seems to stick in octoprint where i set it.

Sebazzz, 2021-01-05:
There is no power loss recovery with Octoprint.

Sebazzz, 2021-01-05:
Note that while the printer is heating up, it is not possible to set the temperature until the printer has heated.

Grimmvg, 2021-01-05:
mm go recheck this when this print is ready. 
I got the impression that i load a file & "print"  , the temp just stays at 0°C .Also the red line in octoprint (actual temp) stays flat.. 
The timer starts running but as long as the extruder is not on temperature, it stays idle. 

Kinda guessed octoprint didn't resume printing or anything , it wasn't working properly on the stock firmware + SD anyway.
Would be a nice feature, would love to sometimes just turn the printer off in the evening and resume the next day

Sebazzz, 2021-01-05:
[image]

[image]

@Grimmvg I see nothing strange here. What stays zero is the offset. The red line stays flat because the bed heats first. That is what your gcode says.

Sebazzz, 2021-01-07:
@Grimmvg Do you encounter the issue still?

Grimmvg, 2021-01-08:
Yes i do , sorry busy days and i'm not everyday in my studio . Also some other things are doing weird like setting the Z offset 

Here are 2 videos of the issue on the printer :

 https://drive.google.com/file/d/10zk7x3PbHKtrVPZEpL2mS70rf1lPnUvf/view?usp=sharing 

https://drive.google.com/file/d/10z39-_nWVK44eIlggunGvvtf2-p2mMqB/view?usp=sharing 

If it would help, i'm happy to do some tests with you over a video call this weekend (i also speak Dutch btw) 

Ps side note : Homing fails quite often (yellow error screen since this update, today i started the printer, preheated, homing , auto bed leveling, started print with octoprint , before the first layer started = homing failed )

Grimmvg, 2021-01-08:
Made a screen capture of octoprint with the issue. 
https://drive.google.com/file/d/1bG1x472ejIi4oM7JTJnjxdgz3FmsO1i1/view?usp=sharing

Also something i forgot to mention before , my Z offset value is not being stored, i have to set it every time i start the printer.
My printer is on a power block with a switch, so i turn the switch on / off to turn the printer on/off as it has the faulty switch and i just bypassed it, didn't take time to order replacement switch yet

Grimmvg, 2021-01-08:
One more weird issue, not sure if it's all related so if i should start a new topic here , but here is a video off it. 
I can't move the X & Y axis below the 0 point.
https://drive.google.com/file/d/11HlxDkpPMQFy0XoHYp6MY1R6XgwzeJor/view?usp=sharing

Sebazzz, 2021-01-08:
I suggest reflashing your firmware altogether (rename the file to be sure). It seems your printer has lost track of its position. Because indeed you should not be able to move them to a position less than zero.

If you pressed disable steppers earlier the numbers should be read. 

Check also issue #127 - a user had still the alpha 4 firmware flashed.

Grimmvg, 2021-01-08:
ok i will try. You mean renaming the firmware file ? 
I can't check it now as it's in a print, but if i recall correctly , it did show the latest update number on the printer info screen.

Grimmvg, 2021-01-08:
the screen says community firmware v2.0.7.1,  guess that's not what it should be then ? I'll anyway try the reflashing with another SD card, hopefully that solves it and i was just another dumb user :D

dryphi, 2021-01-08:
> the screen says community firmware v2.0.7.1, guess that's not what it should be then ? I'll anyway try the reflashing with another SD card, hopefully that solves it and i was just another dumb user :D

Looks like you're still using an older version. The latest release is based on a newer version of Marlin (2.0.7.3 I believe).
https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.3-cr6-community-release-5-beta

Sebazzz, 2021-01-08:
No, it should say CR6Community-Rel5 😀

Grimmvg, 2021-01-08:
omg stupid me :" woo the screens looks new , this update went well" . Not.

Grimmvg, 2021-01-08:
Ok , updated with another SD card , now it does show the Rel5 version in the printer info. Also octorprint seems to be more lively in receiving printer notifications. Bed levelling was indeed slower so guess the update went ok.
First impression is all the issues are solved :D Let me do some more tests but confident this is solved. 
Bit embarrassed, thats all :p

Sebazzz, 2021-01-08:
No problem. I'Il close this now, if there are any issues please open individual issues for them 🙂

closed 2021-01-08 by Sebazzz