CR6Community/Marlin #127: Failure to home after bed leveling

MericasMeuth, 2021-01-06:
*Description of the bug or requested feature*

Bug: After leveling bed, homing fails

*Steps to reproduce*

1. Initiate bed leveling, wait until complete
2. Once bed leveling is complete, manually select auto-home

**Expected behavior:** 

Printer successfully completes auto-bed leveling, then successfully homes

**Actual behavior:** 

Printer successfully completes auto-bed leveling, then fails to home

**Logging:**

I do not have a rasberry pi for octoprint and I could not figure out how to install it on windows

*Additional Information*

Hardware configuration: Stock
Which mainboard: 4.5.2 
Custom mods: None

Video:

https://user-images.githubusercontent.com/72102612/103721063-c2ff4800-4f9a-11eb-87d7-740652af335a.mp4

Exact Firmware Link: ( https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip )

Sebazzz, 2021-01-06:
Please collect logs with Pronterface (included in PrusaSlicer). Also turn off the printer, turn it on again, and use a kitchen scale under the nozzle and lower the nozzle manually (using the couplers) until the blue surface mounted LED on the hot-end PCB comes on. Let's see how many grams it takes (please use metric units).

MericasMeuth, 2021-01-06:
Ok, I'll work on getting this done today/tomorrow.

MericasMeuth, 2021-01-07:
Blue LED comes on at ~127g. I have Pronterface connected to my printer, how do I collect a log for you?

Sebazzz, 2021-01-07:
Instructions are [in the issue template](https://github.com/CR6Community/Marlin/blob/extui/.github/issue_template.md) 😉

To be sure, can you show a picture of the hot-end with its bowden tube and cable?

MericasMeuth, 2021-01-07:
Like with the shroud off?

Sebazzz, 2021-01-07:
No, shroud can stay on

MericasMeuth, 2021-01-07:
[image]
[image]
[image]
[image]
[image]
[image]

MericasMeuth, 2021-01-07:
would you like me to post log from the time i start bed leveling to the time it fails after attempting to home?

Sebazzz, 2021-01-07:
Yes, please, and if possible correlate the lines in the log at what you saw happening at your printer.

---

Stock cable clips, so that pretty much rules out that issue. My strain gauge turns on at at around 150-170g so that is a tiny bit more.

I won't ask you to retry without bowden tube or turning the screw labeled RP1 a hair, because I'm of the opinion that this should work anyway and I assume you didn't have issues on stock firmware. (Which? 1.0.2 or higher)

MericasMeuth, 2021-01-07:
I will do my best on the log... its all a bit confusing.

Community firmware is first time I realized the issue. It was present in the previous release as well.

**Start of Log:**

_Auto-Bed Leveling_
 X88.33 Y30.00 Z-2.20
  current_position= X88.33 Y30.00 Z-0.35 : sync_plan_position
<<< Probe::probe_down_to_z  X88.33 Y30.00 Z-0.35
1st Probe Z:-0.35
>>> do_blocking_move_to  X88.33 Y30.00 Z-0.35
>  X88.33 Y30.00 Z1.65
<<< do_blocking_move_to  X88.33 Y30.00 Z1.65
FIX_MOUNTED_PROBE: Taring the probe
FIX_MOUNTED_PROBE: Taring the probe
FIX_MOUNTED_PROBE: Taring the probe
>>> Probe::probe_down_to_z  X88.33 Y30.00 Z1.65
>>> do_blocking_move_to  X88.33 Y30.00 Z1.65
>  X88.33 Y30.00 Z-2.20
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X88.33 Y30.00 Z-2.20
  current_position= X88.33 Y30.00 Z-0.32 : sync_plan_position
<<< Probe::probe_down_to_z  X88.33 Y30.00 Z-0.32
2nd Probe Z:-0.32 Discrepancy:-0.03
<<< Probe::run_z_probe  X88.33 Y30.00 Z-0.32
>>> do_blocking_move_to  X88.33 Y30.00 Z-0.32
>  X88.33 Y30.00 Z2.67
<<< do_blocking_move_to  X88.33 Y30.00 Z2.67
<<< Probe::probe_at_point  X88.33 Y30.00 Z2.67
>>> Probe::probe_at_point  X88.33 Y30.00 Z2.67
...(146.67, 30.00, raise, 0, probe_relative)
  current_position= X88.33 Y30.00 Z2.67 :
FIX_MOUNTED_PROBE: Disabling probe
>>> do_blocking_move_to  X88.33 Y30.00 Z2.67
>  X146.67 Y30.00 Z2.67
echo:busy: processing
<<< do_blocking_move_to  X146.67 Y30.00 Z2.67
  current_position= X146.67 Y30.00 Z2.67 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X146.67 Y30.00 Z2.67
FIX_MOUNTED_PROBE: Taring the probe
>>> Probe::probe_down_to_z  X146.67 Y30.00 Z2.67
>>> do_blocking_move_to  X146.67 Y30.00 Z2.67
>  X146.67 Y30.00 Z-2.20
echo:busy: processing
<<< do_blocking_move_to  X146.67 Y30.00 Z-2.20
  current_position= X146.67 Y30.00 Z-0.41 : sync_plan_position
<<< Probe::probe_down_to_z  X146.67 Y30.00 Z-0.41
1st Probe Z:-0.41
>>> do_blocking_move_to  X146.67 Y30.00 Z-0.41
>  X146.67 Y30.00 Z1.59
<<< do_blocking_move_to  X146.67 Y30.00 Z1.59
FIX_MOUNTED_PROBE: Taring the probe
FIX_MOUNTED_PROBE: Taring the probe
FIX_MOUNTED_PROBE: Taring the probe
>>> Probe::probe_down_to_z  X146.67 Y30.00 Z1.59
>>> do_blocking_move_to  X146.67 Y30.00 Z1.59
>  X146.67 Y30.00 Z-2.20
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X146.67 Y30.00 Z-2.20
  current_position= X146.67 Y30.00 Z-0.39 : sync_plan_position
<<< Probe::probe_down_to_z  X146.67 Y30.00 Z-0.39
2nd Probe Z:-0.39 Discrepancy:-0.02
<<< Probe::run_z_probe  X146.67 Y30.00 Z-0.39
>>> do_blocking_move_to  X146.67 Y30.00 Z-0.39
>  X146.67 Y30.00 Z2.61
<<< do_blocking_move_to  X146.67 Y30.00 Z2.61
[... 421 lines cut ...]
Move Away:
>>> do_homing_move  X117.00 Y117.00 Z15.00
...(Z, 3.00, 2.00)
  current_position= X117.00 Y117.00 Z15.00 : Probe::set_deployed
deploy: 1
Is in probing zone: 1
Opto switch says: 1
FIX_MOUNTED_PROBE: Preparing for next probe tare
echo:busy: processing
<<< do_homing_move  X117.00 Y117.00 Z15.00
Home 2 Slow:
>>> do_homing_move  X117.00 Y117.00 Z15.00
...(Z, -6.00, 1.00)
  current_position= X117.00 Y117.00 Z15.00 : Probe::set_deployed
deploy: 0
Is in probing zone: 0
Opto switch says: 0
FIX_MOUNTED_PROBE: Taring probe
echo:busy: processing
echo:busy: processing
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

//action:poweroff
**End of Log**

Sebazzz, 2021-01-07:
Is it possible you have still the alpha 4 firmware installed? You can verify this in the info screen.

Met vriendelijke groet,
Sebastiaan Dammann

MericasMeuth, 2021-01-07:
[image]

Sebazzz, 2021-01-07:
That explains the FIX_MOUNTED_PROBE messages in the log. You've installed the community alpha 4 firmware.

Met vriendelijke groet,
Sebastiaan Dammann

[PXL_20210107_171649097 MP]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F72102612%2F103922963-7faff100-50e2-11eb-8937-edcc4e8c4d86.jpg&data=04%7C01%7C%7C0a353a27c1b745f6707408d8b3306275%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637456367671859304%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=peUNFC7uCArhj4aMkQ%2FRH38f7YXR2VDzEBOgg4pYGXc%3D&reserved=0>

MericasMeuth, 2021-01-07:
Well dont I feel stupid... Let me reflash with Beta and see if its fixed.

Sebazzz, 2021-01-07:
No worries. Sometimes the bootloader, especially of v4.5.2 boards, just refuses to update or does only a partial update causing an apparent brick like in #109. If you previously attempted flashing of 5 beta, try reflashing with a different file name (make sure you keep the `.bin` extension).

MericasMeuth, 2021-01-07:
Reflashed 5 Beta firmware. Ran ABL and then homed without any issue.

I believe the issue was the specific SD card. Flashed screen firmware no problem but when i went to flash mobo it wouldnt read. Tried new SD card and worked fine.

Sorry for the headache, and thanks for your time.

Sebazzz, 2021-01-07:
Thank you for confirming! I had some apparently strange SD card issues sometimes too, sometimes reformatting is just the thing. 😀

closed 2021-01-07 by Sebazzz