CR6Community/Marlin #128: Increased probing points do not work

nicwilson58, 2021-01-06:
__Issues that do not follow this issue template will be closed__

- *Be sure you have fully read the release notes and flashing instructions before posting an issue here*
- *Check if there is an existing _open_ issue that describes your problem and add your comments there*
- *Check if your issue has been resolved on the latest source code - or if there is a closed issue pointing to the next release*
- *Check that you have flashed the correct firmware for your device*

### Description

*Increased probing points does not work, display changes correctly, probing points do not*

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. change GRID_MAX_POINTS_X to 5
2. run bed leveling
3. 

**Expected behavior:** 

Bed level should probe 25 points

**Actual behavior:** 

Display correctly changes to 25 points
Bed only probes 16,  and ends when display highlights probe point 16 of 25

this is all it returned

Send: M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
Recv: ok

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

code build from new download unmodified except for number of points change

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

Standard unmodified.

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

they were unmodified except for GRID_MAX_POINTS_X fronm 4 to 5

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
N/A  compiled from your source code 6/1/21

Sebazzz, 2021-01-06:
Are you positive your firmware got actually flashed? It is impossible for bed leveling to *terminate* after 16 points, but it is certainly so that we can only show a maximum of 16 points and that will not change.

nicwilson58, 2021-01-06:
Hi,

 

You are not understand, and you reply is incorrect. The graphic display DOES show more than 16,   its shows 25 on the display, but the bed only goes through 16,  so this is not impossible.  Please try it for yourself,  change it to 5 compile it and watch.   If it had not flashed,  how did the graphic display change to 25?

 

Nic

nicwilson58, 2021-01-06:
You can show more than 16 points,  so your comment there is incorrect.   for a grid max of 5 your display shows 49 points,  the bed did so 25 when counted them correctly,  the display showed 25 of 49 when it ended.  I will look into changing the display to show the correct number of points myself on my branch.

closed 2021-01-06 by nicwilson58

Sebazzz, 2021-01-06:
Yes, only the default probing points of the MAX and SE are supported, though it is quite easy to swap in a new ICL with the correct icons.

nicwilson58, 2021-01-06:
Hi,

 

Well yes and  no.   the bed changes to the new point number,  and the display changes to a wrong number (49) so your comment that only the default probing points are supported is not accurate.  If that was so the display would remain on 16,  or the bed prevented from going over 16.  As it stands its wrong and confusing for any user compiling his own and expecting the points to change.  But its your code,  you can keep it any way you want to keep it.   J

 

Cheers