CR6Community/Marlin #131: [FR] Please consider helping new users avoid raising repeat Issues

Thinkersbluff, 2021-01-06:
I realize there will always be some users who flag known issues (e.g. "positions are displayed in cm instead of mm"), but as this firmware becomes increasingly popular with first-time users who may not be familiar with GitHub, Discord, detailed Release documentation, software release cycles, etc., I suspect the usefulness of this Issues section is going to be significantly diminished by avoidable "noise" from well-meaning users who "don't know if you know, but..." or who are confused by or in disagreement with design decisions over which there has already been significant and decisive discussion.

Your experience in this domain may suggest better options than I.

My suggestions - for your consideration- are:
- Please adopt a philosophy of not closing Issue reports until the release that fixes them is released.
  That way, users can at least scan existing issues before raising redundant ones.  You can still use tags to flag & filter on issues you consider "resolved" & if you develop a release plan, you could tag issues with the targeted release#.
- Please update the "Known Issues" section of the current Release documentation, if one or more commits is required to resolve an issue but you decide not to release the new build, yet.  (Users who use only pre-compiled firmwares are most likely to be flagging issues you have already resolved.  It might help both them and the devs, if they could find confirmations and work around advice in the release notes, to tide them over until the next build is posted.) 
- Please consider modifying the ISSUE template to call for a cross-reference to any "similar" issues already open or closed, to increase the probability that well-meaning users find the help they need without having to open redundant new issues.

Cheers.

Sebazzz, 2021-01-06:
Thank you for the feedback - I've updated the release notes and issue template with a reference to the [community firmware release 6 milestone](https://github.com/CR6Community/Marlin/milestone/3?closed=1).

I will however close issues that are resolved in source code, otherwise I lose track which things need attention and which do not. 
(cc @CR6Community/developers )

closed 2021-01-06 by Sebazzz

Thinkersbluff, 2021-01-06:
Thanks for considering the feedback.  
I see the documentation update.  Thank you.  
That should help, for those who read documentation.  (We "process weenies" are a lonely under-appreciated lot...  😬 )