CR6Community/Marlin #132: Bed Temp (80° for PETG) was lowered to 50° after homing

krammerj2486, 2021-01-10:
### Description

After starting printing via Octoprint, the bed temperature was suddenly lowered from 80° to 50° after homing takes action. 
  
### Steps to Reproduce

1. preheat with Octoprint with PLA temp settings (in my case 200/60)
2. wait until target temp is reached (not longer)
3. start print job (PETG with 240/80)
4. the printer is heating to target temp (first bed, then nozzle)
5. then "Heating up and homing in progress" is displayed
6. While homing bed target temp is lowered to 50° and sticks there while printing. Nozzle temp heats up to 240° as desired

I think homing is done with 120°/50°. While the nozzle goes back to 240, the bed is still 50°
first T message after homing   -> Recv:  T:234.81 /120.00 B:79.85 /50.00 @:0 B@:0
next T message                         -> Recv:  T:226.00 /240.00 B:78.93 /50.00 @:0 B@:0 W:?

after setting it manually to 80 -> Recv:  T:219.53 /240.00 B:78.03 /80.00 @:127 B@:127 W:?

**Expected behavior:** 

G-Code defined bed temp is used (80°)

**Actual behavior:** 

Print job is done with a bed temp of 50° instead of 80°. Homing set temps to 120/50, but didn't set the bed temp back to 80°. Manuel change to 80° is still working. 

**Logging:**

Recv:  T:230.12 /240.00 B:79.87 /80.00 @:96 B@:64 W:?
Recv: echo:busy: processing
Recv:  T:230.94 /240.00 B:79.89 /80.00 @:83 B@:60
Recv: Hotend Idle Timeout
Recv: //action:notification Hotend Idle Timeout
Recv: //action:notification Printing...
Recv: ok
Send: N48 M117 ETA 1/09 8.34PM*115
Recv: //action:notification ETA 1/09 8.34PM
Recv: ok
Send: N49 M73 P0*106
Recv: ok
Send: N50 M117 Haltebock_Kabelschlepp_00860003_0.2mm_PETG_CR6SE_10h36m.gcode*3
Recv: //action:notification Haltebock_Kabelschlepp_00860003_0.2mm_PETG_CR6SE_10h36m.gcode
Recv: ok
Send: N51 M73 P0*99
Recv: ok
Send: N52 M117 0.01% complete*32
Recv: //action:notification 0.01% complete
Recv: ok
Send: N53 M73 P0*97
Recv: ok
Send: N54 M117 ETL 10h39m54s*52
Recv: //action:notification ETL 10h39m54s
Recv: ok
Send: N55 M73 P0*103
Recv: ok
Send: N56 M117 ETA 1/09 8.35PM*125
Recv: //action:notification ETA 1/09 8.35PM
Recv: ok
Send: N57 M73 P0*101
Recv: ok
Send: N58 M117 Haltebock_Kabelschlepp_00860003_0.2mm_PETG_CR6SE_10h36m.gcode*11
Recv: //action:notification Haltebock_Kabelschlepp_00860003_0.2mm_PETG_CR6SE_10h36m.gcode
Recv: ok
Send: N59 M73 P0*107
Recv: ok
Send: N60 M117 0.01% complete*33
Recv: //action:notification 0.01% complete
Recv: ok
Send: N61 M73 P0*96
Recv: ok
Send: N62 M117 ETL 10h39m54s*49
Recv: //action:notification ETL 10h39m54s
Recv: ok
Send: N63 M73 P0*98
Recv: ok
Send: N64 G90*34
Recv: ok
Send: N65 G92 E0*116
Recv: X:117.00 Y:117.00 Z:10.00 E:0.00 Count X:9360 Y:9360 Z:4000
Recv: ok
Send: N66 G1 Z2.0 F3000*43
Recv: ok
Send: N67 G1 X10.1 Y20 Z0.28 F5000.0*23
Recv:  T:233.00 /120.00 B:79.85 /50.00 @:0 B@:0
Recv: ok
Send: N68 G1 X10.1 Y200.0 Z0.28 F1500.0 E15*86
Recv:  T:234.81 /120.00 B:79.85 /50.00 @:0 B@:0
Recv: echo:busy: processing
Recv: ok
Send: N69 G1 X10.4 Y200.0 Z0.28 F5000.0*50
Recv: ok
Send: N70 G1 X10.4 Y20 Z0.28 F1500.0 E30*115
Recv:  T:234.75 /120.00 B:79.59 /50.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:233.00 /120.00 B:79.41 /50.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:231.00 /120.00 B:79.24 /50.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:228.87 /120.00 B:79.09 /50.00 @:0 B@:0
Recv: ok
Send: N71 M117 ETA 1/09 8.34PM*121
Recv: //action:notification ETA 1/09 8.34PM
Recv: ok
Send: N72 M73 P0*98
Recv: ok
Send: N73 G92 E0*115
Recv: X:10.40 Y:20.00 Z:0.28 E:0.00 Count X:832 Y:7200 Z:51
Recv: ok
Send: N74 G1 Z2.0 F3000*40
Recv: ok
Send: N75 M75*35
Recv: ok
Send: N76 M109 S240*95
Recv:  T:226.00 /240.00 B:78.93 /50.00 @:0 B@:0 W:?
Recv:  T:225.75 /240.00 B:78.90 /50.00 @:127 B@:0
Recv:  T:224.96 /240.00 B:78.80 /50.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:223.57 /240.00 B:78.79 /50.00 @:127 B@:0 W:?
Recv:  T:222.90 /240.00 B:78.66 /50.00 @:127 B@:0
Recv:  T:222.63 /240.00 B:78.64 /50.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:221.43 /240.00 B:78.49 /50.00 @:127 B@:0 W:?
Recv:  T:220.76 /240.00 B:78.48 /50.00 @:127 B@:0
Recv:  T:220.67 /240.00 B:78.37 /50.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:219.90 /240.00 B:78.33 /50.00 @:127 B@:0 W:?
Recv:  T:220.00 /240.00 B:78.27 /50.00 @:127 B@:0
Recv:  T:219.48 /240.00 B:78.18 /50.00 @:127 B@:0 W:?
Recv: echo:busy: processing
Recv:  T:219.17 /240.00 B:78.08 /50.00 @:127 B@:0 W:?
Recv:  T:219.17 /240.00 B:78.03 /50.00 @:127 B@:0
Recv:  T:219.53 /240.00 B:78.03 /80.00 @:127 B@:127 W:?
Recv: echo:busy: processing

### Additional Information

Hardware is standard CR6-SE with stock mainboard 4.5.2  

Used firmware:
https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

[image]
[image]
[image]
[image]

Sebazzz, 2021-01-10:
Hi @krammerj2486 

Thank you for your report. I think this might be fixed already on the latest sources.

Two things you can try:
- In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try the bugfix ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)).
- Otherwise, put the temperature commands [after the G28 command like this](https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping) until an official release is made.

ValChe82, 2021-01-12:
This bug appears when the printer hit the standby mode during the homing before printing.

Sebazzz, 2021-01-26:
@ValChe82 I agree - I don't see this happening anymore and the heating is quite slow so you're more likely to hit the 5 minute timeout. 

Assuming this is fixed I will close this now (I did a bunch of PETG prints for my H2 extruder mount and I didn't notice anything).

closed 2021-01-26 by Sebazzz