CR6Community/Marlin #134: "Left line" is printed with 120° after homing - cold extrusion

krammerj2486, 2021-01-10:
### Description

Homing is done with a nozzle temp of 120° and the bed temp from the G-Code (in my Case 80°). After Homing the line on the left side is "printed" with 120° nozzle temp. Then it heats up to the target nozzle temp (in my case 240°) and starts printing the skirt. Then it worked fine, but the line on the left side is missing.

### Steps to Reproduce
1. Starting a PETG print job from OctoPrint, with 240/80 as a target temp. No preheating on the printer or OctoPrint is used! 
2. Printer starts to preheat the Bed to 80°
3. Then the "Heating up and homing in progress" message ist shown on display
4. Nozzle is heated to 120°
5. Homing takes action, Nozzle heating is switched off while probing
6. After probing the nozzle is heated to 120°
7. and started to print the left line with 120° nozzle temp and 80° bed temp
8. Then stopped at the left front corner and heated to the G-Code target Temp 240°
9. printing the Skirt with correct temp setting

Recv:  T:118.93 /120.00 B:80.00 /80.00 @:48 B@:54
Recv: ok
Send: N59 G1 X10.1 Y20 Z0.28 F5000.0*26
Recv: ok
Send: N60 G1 X10.1 Y200.0 Z0.28 F1500.0 E15*94
Recv: echo: **cold extrusion prevented**
Recv:  T:118.86 /120.00 B:80.00 /80.00 @:46 B@:55
**Expected behavior:** 

printing the left line with G-Code target temp instead of 120° (probing temp) 

**Actual behavior:** 

Line is "printed" with 120° s for probing.

**Logging:**

Recv:  T:123.86 /120.00 B:80.55 /80.00 @:0 B@:45 W:9
Recv: echo:busy: processing
Recv:  T:123.97 /120.00 B:80.50 /80.00 @:0 B@:52
Recv:  T:124.08 /120.00 B:80.44 /80.00 @:0 B@:59 W:9
Recv:  T:124.12 /120.00 B:80.43 /80.00 @:0 B@:55 W:9
Recv: echo:busy: processing
Recv:  T:124.10 /120.00 B:80.43 /80.00 @:0 B@:53
Recv:  T:123.97 /120.00 B:80.43 /80.00 @:0 B@:51 W:9
Recv:  T:123.81 /120.00 B:80.43 /80.00 @:0 B@:46 W:9
Recv: echo:busy: processing
Recv:  T:123.69 /120.00 B:80.41 /80.00 @:0 B@:48
Recv:  T:123.53 /120.00 B:80.42 /80.00 @:0 B@:45 W:9
Recv:  T:123.20 /120.00 B:80.34 /80.00 @:0 B@:54 W:9
Recv: echo:busy: processing
Recv:  T:122.95 /120.00 B:80.29 /80.00 @:0 B@:58
Recv:  T:122.79 /120.00 B:80.29 /80.00 @:0 B@:57 W:9
Recv:  T:122.26 /120.00 B:80.29 /80.00 @:0 B@:52 W:8
Recv: echo:busy: processing
Recv:  T:122.07 /120.00 B:80.29 /80.00 @:0 B@:51
Recv:  T:121.76 /120.00 B:80.29 /80.00 @:0 B@:48 W:7
Recv:  T:121.22 /120.00 B:80.29 /80.00 @:1 B@:46 W:6
Recv: echo:busy: processing
Recv:  T:120.97 /120.00 B:80.29 /80.00 @:4 B@:44
Recv:  T:120.69 /120.00 B:80.21 /80.00 @:8 B@:54 W:5
Recv:  T:120.00 /120.00 B:80.15 /80.00 @:17 B@:60 W:4
Recv: echo:busy: processing
Recv:  T:119.84 /120.00 B:80.17 /80.00 @:18 B@:56
Recv:  T:119.44 /120.00 B:80.14 /80.00 @:23 B@:57 W:3
Recv:  T:118.86 /120.00 B:80.14 /80.00 @:31 B@:53 W:2
Recv: echo:busy: processing
Recv:  T:118.57 /120.00 B:80.14 /80.00 @:34 B@:52
Recv:  T:118.31 /120.00 B:80.14 /80.00 @:37 B@:50 W:1
Recv:  T:117.72 /120.00 B:80.14 /80.00 @:44 B@:48 W:0
Recv: echo:busy: processing
Recv: //action:notification Printing...
Recv: Taring probe
Recv:  T:117.51 /0.00 B:80.12 /80.00 @:0 B@:51
Recv: echo:busy: processing
Recv:  T:116.58 /0.00 B:80.03 /80.00 @:0 B@:59
Recv: echo:busy: processing
Recv:  T:116.00 /0.00 B:80.00 /80.00 @:0 B@:57
Recv: echo:busy: processing
Recv:  T:115.32 /0.00 B:79.99 /80.00 @:0 B@:54
Recv: echo:busy: processing
Recv:  T:114.51 /0.00 B:80.00 /80.00 @:0 B@:50
Recv: echo:busy: processing
Recv:  T:113.59 /120.00 B:79.97 /80.00 @:74 B@:53
Recv: echo:busy: processing
Recv:  T:112.64 /120.00 B:79.89 /80.00 @:98 B@:63
Recv: echo:busy: processing
Recv:  T:112.26 /0.00 B:79.87 /80.00 @:0 B@:62
Recv: echo:busy: processing
Recv:  T:112.36 /0.00 B:79.85 /80.00 @:0 B@:62
Recv: echo:busy: processing
Recv:  T:112.37 /0.00 B:79.87 /80.00 @:0 B@:58
Recv: echo:busy: processing
Recv:  T:112.09 /120.00 B:79.88 /80.00 @:83 B@:57
Recv: echo:busy: processing
Recv:  T:111.65 /120.00 B:79.92 /80.00 @:101 B@:51
Recv: echo:busy: processing
Recv:  T:111.53 /120.00 B:79.94 /80.00 @:112 B@:49
Recv: SetNewScreen: 37
Recv: X:117.00 Y:117.00 Z:10.00 E:0.00 Count X:9360 Y:9360 Z:4000
Recv: Waiting to heat-up again before continueing
Recv: //action:notification Waiting for heat-up...
Recv:  T:111.64 /120.00 B:79.91 /80.00 @:112 B@:56 W:?
Recv: echo:busy: processing
.
.
.
Recv:  T:119.44 /120.00 B:80.00 /80.00 @:46 B@:54 W:1
Recv:  T:119.29 /120.00 B:79.99 /80.00 @:47 B@:55
Recv:  T:119.20 /120.00 B:80.00 /80.00 @:47 B@:54 W:0
Recv: echo:busy: processing
Recv: //action:notification Printing...
Recv: ok
Send: N34 M117 ETL 10h04m58s*48
Recv: //action:notification ETL 10h04m58s
Recv: ok
Send: N35 M73 P0*97
Recv: ok
Send: N36 M117 ETA 1/10 7.48PM*118
Recv: //action:notification ETA 1/10 7.48PM
Recv: ok
Send: N37 M73 P0*99
Recv: ok
Send: N38 M117 Ausgleichsplatte_00860015_0.2mm_PETG_CR6SE_12h43m.gcode*20
Recv: //action:notification Ausgleichsplatte_00860015_0.2mm_PETG_CR6SE_12h43m.gcode
Recv: ok
Send: N39 M73 P0*109
Recv: ok
Send: N40 M117 0.01% complete*35
Recv: //action:notification 0.01% complete
Recv: ok
Send: N41 M73 P0*98
Recv: ok
Send: N42 M117 ETL 10h04m58s*49
Recv: //action:notification ETL 10h04m58s
Recv: ok
Send: N43 M73 P0*96
Recv: ok
Send: N44 M117 ETA 1/10 7.48PM*115
Recv: //action:notification ETA 1/10 7.48PM
Recv: ok
Send: N45 M73 P0*102
Recv: ok
Send: N46 M117 Ausgleichsplatte_00860015_0.2mm_PETG_CR6SE_12h43m.gcode*29
Recv: //action:notification Ausgleichsplatte_00860015_0.2mm_PETG_CR6SE_12h43m.gcode
Recv: ok
Send: N47 M73 P0*100
Recv: ok
Send: N48 M117 0.01% complete*43
Recv: //action:notification 0.01% complete
Recv: ok
Send: N49 M73 P0*106
Recv: ok
Send: N50 M117 ETL 10h04m58s*50
Recv: //action:notification ETL 10h04m58s
Recv: ok
Send: N51 M73 P0*99
Recv: ok
Send: N52 M117 ETA 1/10 7.49PM*117
Recv: //action:notification ETA 1/10 7.49PM
Recv: ok
Send: N53 M73 P0*97
Recv: ok
Send: N54 M117 Ausgleichsplatte_00860015_0.2mm_PETG_CR6SE_12h43m.gcode*30
Recv: //action:notification Ausgleichsplatte_00860015_0.2mm_PETG_CR6SE_12h43m.gcode
Recv: ok
Send: N55 M73 P0*103
Recv: ok
Send: N56 G90*35
Recv: ok
Send: N57 G92 E0*117
Recv: X:117.00 Y:117.00 Z:10.00 E:0.00 Count X:9360 Y:9360 Z:4000
Recv: ok
Send: N58 G1 Z2.0 F3000*38
**Recv:  T:118.93 /120.00 B:80.00 /80.00 @:48 B@:54
Recv: ok
Send: N59 G1 X10.1 Y20 Z0.28 F5000.0*26
Recv: ok
Send: N60 G1 X10.1 Y200.0 Z0.28 F1500.0 E15*94
Recv: echo: cold extrusion prevented
Recv:  T:118.86 /120.00 B:80.00 /80.00 @:46 B@:55**
Recv: echo:busy: processing
Recv:  T:118.99 /120.00 B:79.99 /80.00 @:42 B@:56
Recv: echo:busy: processing
Recv:  T:119.29 /120.00 B:80.00 /80.00 @:37 B@:55
Recv: ok
Send: N61 M117 0.01% complete*32
Recv: //action:notification 0.01% complete
Recv: ok
Send: N62 M73 P0*99
Recv: ok
Send: N63 G1 X10.4 Y200.0 Z0.28 F5000.0*56
Recv: ok
Send: N64 G1 X10.4 Y20 Z0.28 F1500.0 E30*118
Recv: echo: cold extrusion prevented
Recv:  T:119.71 /120.00 B:80.00 /80.00 @:31 B@:55
Recv: echo:busy: processing
Recv:  T:120.15 /120.00 B:80.00 /80.00 @:25 B@:55
Recv: echo:busy: processing
Recv:  T:120.44 /120.00 B:80.00 /80.00 @:22 B@:55
Recv: echo:busy: processing
Recv:  T:120.63 /120.00 B:80.00 /80.00 @:20 B@:55
Recv: ok
Send: N65 G92 E0*116
Recv: X:10.40 Y:20.00 Z:0.28 E:0.00 Count X:832 Y:7200 Z:51
Recv: ok
Send: N66 G1 Z2.0 F3000*43
Recv: ok
Send: N67 M75*32
Recv: ok
Send: N68 M109 S240*80
Recv:  T:120.72 /240.00 B:80.00 /80.00 @:19 B@:55 W:?
### Additional Information

Printer is CR-6 SE with stock Mainboard 4.5.2
Firmware:
https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

Sebazzz, 2021-01-10:
> Recv: SetNewScreen: 37
> Recv: X:117.00 Y:117.00 Z:10.00 E:0.00 Count X:9360 Y:9360 Z:4000
> Recv: Waiting to heat-up again before continueing
> Recv: //action:notification Waiting for heat-up...
> Recv: T:111.64 /120.00 B:79.91 /80.00 @:112 B@:56 W:?

It heats up to the previously set temperature. This suggests T=120C was previously set. Can I have a part of your gcode? (The first layer)

Two things you can try:
- In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try the bugfix ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)).
- Otherwise, put the temperature commands [after the G28 command like this](https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping) until an official release is made.

krammerj2486, 2021-01-10:
I'll try the new build after my print is finished. Find the GCode snipped attached to this message...

[Testfile_firstLayer_0.2mm_PETG_CR6SE.txt](https://github.com/CR6Community/Marlin/files/5792502/Testfile_firstLayer_0.2mm_PETG_CR6SE.txt)

krammerj2486, 2021-01-10:
I don't think that it is heating to the previously set temp, because the 120° came from the homing process
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Sebazzz, 2021-01-16:
Is that with the prerelease firmware @krammerj2486 - you should see "CR6-Release6-Pre1" in the about menu.

krammerj2486, 2021-01-17:
Now I tried it with the prerelease 6.0. The problem is still there, but know it tries to print the purge line with 145°. In the middle of the second purge line it switches to 240°.

Sebazzz, 2021-01-17:
Do you have a minimally reproducible gcode file and exact reproduction steps? (I can’t check from my cellphone if you did already put it in the issue)

Met vriendelijke groet,
Sebastiaan Dammann

krammerj2486, 2021-01-17:
Yes GCode snipped and steps are in the issue. But I think it should happend with any g-code file that has a purge line in it

krammerj2486, 2021-01-17:
[TestGCodeFile.txt](https://github.com/CR6Community/Marlin/files/5825928/TestGCodeFile.txt)

I'm not able to upload G-Code here, hope the txt helps. The steps are in the issue

Sebazzz, 2021-01-17:
I'm reviewing your gcode @krammerj2486 but I see something odd:

```gcode
M107
M190 S80 ; set bed temperature and wait for it to be reached
M104 S240 ; set temperature
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G90 ;Absolute positionning
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

M75 ;start print timer
M109 S240 ; set temperature and wait for it to be reached
```

You have M109 after drawing the purge line. You have M104 earlier but that sets the temperature but it does not wait for it. After homing it of course immediately proceeds drawing the purge line, but at the temperature set because. But it homes immediately, turning off the heater. 

Am I wrong in stating this isn't necessarily a bug? You instructed the printer to only wait for the temperature *after* drawing the purge line.

krammerj2486, 2021-01-17:
You are right, that comes somehow from PrusaSlicer, in Cura generate always M140 immediately followed by M190 and 104 followed by 109. Strange I have to take a look what is wrong in PS setting.

M140 S65
M105
M190 S65
M104 S210
M105
M109 S210

krammerj2486, 2021-01-17:
I took a look at prusaSlicer but I found no setting that could influence the "wait for nozzle temp". The start code seems ok there are no temp settings in it, but the slicer inserts the wrong temp gcode (or the code to wait for the nozzle temp is missing)

Sebazzz, 2021-01-18:
Can you show what start gcode you are using in PrusaSlicer?

krammerj2486, 2021-01-18:
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
G28 ;Home
G90 ;Absolute positionning
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
M75 ;start print timer

Sebazzz, 2021-01-18:
Unless I'm missing something, you should add a `M109 S[first_layer_temperature]` in your start gcode. Please also see the comment from the PrusaSlicer devs here: https://github.com/prusa3d/PrusaSlicer/issues/2961

Sebazzz, 2021-01-26:
May I assume this is purely a start gcode thing? That you didn't run into it in the previous firmware is because the heater wasn't paused while probing. The gcode was always incorrect, but now it actually shows 😉

Feel free to reopen if I'm incorrect.

closed 2021-01-26 by Sebazzz

krammerj2486, 2021-01-29:
where would you place the "M109 S[first_layer_temperature]"? I suggest after the G92 E0, is this OK? You wrote "unless I#m missing something" - what are you missing?

Sebazzz, 2021-01-29:
After the `G28` it should be added.

krammerj2486, 2021-01-29:
OK thanks!