CR6Community/Marlin #135: [FR] Please consider adding “help” screens to the UI.

Thinkersbluff, 2021-01-10:
As this UI becomes increasingly feature-rich, and as it attracts an increasing number of new Users, I expect you will have difficulty responding to Users who would rather ask a question than read the release notes...

In the same way that Ghostkeeper’s Cura Add-In provides tips embedded into the Cura slicer settings, it might be helpful to provide “tips” on how to use this UI.  The new esteps function, for instance, warrants a User Help screen telling them how to prepare their machine for the test.  See my feedback on #57 

E.g. 
- A small “?” Icon or “i” in a circle could be placed i the right top corner of a screen for which help was available. 
- Tapping on that icon could open the Help screen.  
- If multiple screens were required, there could be a page function similar to the SD card screens.  
- The Info screen could give access to to full set with a scrollable TOC.

grobux, 2021-01-10:
I agree, but I think it would be easier to complete the primitive website we have for now, with a wiki/FAQ section explaining everything. That would also allow us to add pictures, links, etc.

Repo: https://github.com/CR6Community/cr6community.github.io/
Frontend: https://cr6community.github.io/

Thinkersbluff, 2021-01-10:
I think there is place for both.  
Priorities for your spare time are entirely yours.  Just wanted to plant the seed. 

Personally, I am disappointed by how any Users ask questions already answered in the release notes. 

 I theorize that some of us don’t like to read a whole page to find one piece of info, especially I we can just ask whether anybody already knows...

Maybe we can train some users to check for help screens before posting issues.  Much faster to find and scan, in the heat of the moment.

Sebazzz, 2021-01-10:
> Personally, I am disappointed by how any Users ask questions already answered in the release notes.

Those are the same users who don't read one inch further where their nose ends 😉 We can add all the tooltips we want, and it won't help.

But from the technical side this is probably quite the challenge. I don't think DWIN really has any tooltips or similar.

Thinkersbluff, 2021-01-10:
> I agree, but I think it would be easier to complete the primitive website we have for now, with a wiki/FAQ section explaining everything. That would also allow us to add pictures, videos, links, etc.
> 
> Repo: https://github.com/CR6Community/cr6community.github.io/
> Frontend: https://cr6community.github.io/

Thanks for these links. I had not noticed them before.  Some day, I hope to be able to compile my own copy of the DWIN_SET, but my Norton SafeWeb add-in keeps warning me off the various resources suggested by Google.  I would love some guidance in the FAQ/wiki on how to set up an equivalent to my VSCode/Platformio IDE for compiling the DIsplay firmware.

Thinkersbluff, 2021-01-10:
> from the technical side this is probably quite the challenge. I don't think DWIN really has any tooltips or similar.

I only expected something like your existing menu buttons, paging back and forth between a short note and the menu it supported, not pop up tool tips. Sorry if I was not not clear.

If there is any merit to the idea, I get that the timing may not be right.  Please just salt it away until it appeals as an option.

Sebazzz, 2021-01-10:
If we're going to add some help pages, we're going to need text for it.

Thinkersbluff, 2021-02-03:
It looks like I am on the fringe with this idea.  I suggest we close it until there is a more compelling reason for it.

closed 2021-02-03 by Thinkersbluff

Beeble2695, 2021-02-03:
Could always just put a URL in the "?" help page ;)