CR6Community/Marlin #137: [FR]  Please add an Exit or Cancel button to the "Heating, please wait" screen(s)

Thinkersbluff, 2021-01-10:
"Fat-fingered boob" that I can be, I have several times accidentally hit "HOME" and then wished I hadn't, when it says, "Heating, please wait"  (e.g. bumping the x axis in 10mm increments while watching the printer & not the screen, accidentally hit Home)

If there are other people like me, we would all probably appreciate the chance to change our minds sometimes and Cance/Exit out of that situation, without having to power cycle the printer.

Sebazzz, 2021-01-10:
Sorry, this is not possible. Homing cannot be canceled in Marlin.

Technically it can (M410 - stepper quickstop) but it will cause an attached Octoprint instance to sound the alarm and issue an M112 which will kill the printer.

Thinkersbluff, 2021-01-10:
> Sorry, this is not possible. Homing cannot be canceled in Marlin

Rats.  So, patience, more care, or power-cycle are my only options...

I don't think we all want to go through that "Are you sure?" gatekeeper, which is the only other thought that comes to mind.

grobux, 2021-01-10:
Maybe we can limit miss-clicks by moving the home button out of the axis actions.
We have ergonomic adjustments with higher priority to do before. ;)

Sebazzz, 2021-01-10:
We can decrease the size of the touch sensitive area.

Met vriendelijke groet,
Sebastiaan Dammann

Thinkersbluff, 2021-01-10:
Could we reopen this FR then, please, with a new title?
e.g.: "Please help Users avoid pressing "Home" by accident whien using MOVE commands without looking"?

Sebazzz, 2021-01-10:
Sure!

Met vriendelijke groet,
Sebastiaan Dammann

Allan-N, 2021-01-10:
Is (or should there be) a difference between homing (including hitting the end-stops) vs. just moving to X0, Y0, Z0 ?

Also, should the Home button between the X+/X- and Y+/Y- move buttons just handle X/Y and then we add a Home button between Z+/Z- ?

Thinkersbluff, 2021-01-10:
> Is (or should there be) a difference between homing (including hitting the end-stops) vs. just moving to X0, Y0, Z0 ?
> 
> Also, should the Home button between the X+/X- and Y+/Y- move buttons just handle X/Y and then we add a Home button between Z+/Z- ?

I have mentioned before that I really "miss" my Home X, Home Y and Home Z commands on this printer. (I also have an Ender 3 with Marlin LCD menus.  These three commands are on their Motion menu.)

The endstops are still involved when homing one axis.  I can just command an axis to 0.0 with the existing Move controls, if I don't want to hit the stop, but that is when I am at risk if I hit Home Axis after having manually physically moved that axis on the printer.  That was why I asked for the cue that became red position text.

Sebazzz, 2021-02-04:
Touch sensitive size decreased from 48px to 36px.

closed 2021-02-04 by Sebazzz