CR6Community/Marlin #138: Add PID "BED" tuning directly from printer

Allan-N, 2021-01-10:
### Description

Add an option to perform the PID BED auto-tuning without using a USB connected device (OctoPrint, Pronterface, etc).

### Steps to Reproduce

As a follow-on to #60, we should also provide a way to perform PID BED tuning.

1. Select a PID "BED" tuning menu option
2. Provide some feedback as the PID tuning and PID Bed temperature tuning is progresses
3. Save the newly computed values

**Expected behavior:** 

### Additional Information

All printers need some setup. Why not make things simpler and minimize the need for a connected host.

### Thoughts

One might consider adding a "Hot End" / "Bed" toggle to the PID tuning screen

ThisIsMyUniqueUsername, 2021-02-14:
The screens are already quite cluttered, is it really necessary to add a function that most users will never even touch? How often do you replace your bed heaters?

Sebazzz, 2021-02-14:
Yup, that's the reason I've put a low prio tag on it. But this can be as easy as putting a "Bed/Hotend" toggle on the same screen.

Thinkersbluff, 2021-02-18:
> The screens are already quite cluttered, is it really necessary to add a function that most users will never even touch? How often do you replace your bed heaters?

If a user prints in ABS, PETG, PLA, TPU, etc..  Should they try to make one BED PID work for all materials, or should they PID the BED as part of reconfiguring their printer for the new material they plan to print with?

If PID is better than Bang-Bang for the bed, then I think we should help users who are unwilling or unable to use the USB interface to be able to adjust their bed PID settings.  

As @Sebazzz notes, this need not be a big "UI clutter" thing.  Just a switch on the PID tuning menu.  I still like the idea of integrating the PLA, PETG and Custom pre-heat settings into the PID tuning function. (eg. "Set as...")

Lorraine1000, 2021-04-10:
Is it normal that the bed temperature deviates up to 5°C from the displayed temperature on the touchscreen? In the middle of the bed it is around 5°C higher than displayed (so i.e. 60°C displayed is actually 65°C), lowering to a deviation of 2-3°C at the outer sides (and even going down to -2 to -3° in selected corners of the print bed). All measured with a Voltcraft IR-270L infrared thermometer, ambient temps in the room about ~21°C.

And if it is "normal" (because of the poor hardware design of the CR6 SE), could those temperature differences somehow be considered by the firmware during first-layer print, similar to the bed leveling mesh? Something like e.g. increasing nozzle temp to heat up the material more for the "colder" areas and lower it for the "hotter" areas? 

Not sure whether 2-5°C nozzle temp could be changed swiftly enough - or whether that would make sense at all, but it doesn't hurt to ask at least :-)

tomearp, 2021-04-13:
i can't say i've looked in detail at the spread of heat on the bed before. the layout of the heating element seems uniform enough from looking at it. i'll pick up my thermal imaging camera from the office on friday and see what it looks like.

without measuring anything, i would guess that the distribution of heat could be improved by introducing an insulating material to the underside of the bed. from a quick glance underneath it appears a very thin layer of something has been put over the location of the bed thermistor to try and improve its ability to measure the temperature of the bed without losing too much heat to the surroundings.

i would think the hotend would not be able to respond quickly enough, and i'm not sure compensating with changing nozzle temperature would be the right thing to do. someone else will need to confirm that, i'm way out of my depth there.

@Lorraine1000 are you on the community firmware discord? it might be better to continue the discussion there so we don't hijack this feature request. i'm cowgoesmoo on there.

https://discord.gg/njBanx75