CR6Community/Marlin #139: DGUS_LCD_UI_CREALITY_TOUCH is ambiguous

mrv96, 2021-01-11:
Hi, I open this issue in order to find a solution to avoid Marlin users confusion.

Looking this lines:
https://github.com/CR6Community/Marlin/blob/42916c7896608f91813d8896dc5f1a89a9de4ec8/Marlin/Configuration.h#L2250-L2262
it seems that this four options share the same hardware, but this is not the case.

The three original Marlin options works with T5UID1 display, while yours works with T5L display.

Do you think that is reasonable underline this aspect either using comments or changing the define?

IMHO add a reference to display hardware would be necessary also in official Marlin. Right now there are a lot of similar but different DGUS LCDs

Sebazzz, 2021-01-16:
@grobux I think you found the original AliExpress link where to purchase this?

grobux, 2021-01-16:
I'm experimenting right now with a DWIN screen, reference DMG48270C043_03WTC (item+delivery is around 20€).
The last letter in the reference is for the type of panel : C=capacitive, R=resistive, and N=no touch.

I'm not 100% sure, but the hardware is pretty much the same as the stock screen.
I flashed the Community screen firmware without too much trouble (had to do it from the SD card twice, maybe to correctly update the OS).

commit 1100486e (2021-01-26): Add part number to the DGUS_LCD_UI_CREALITY_TOUCH flag - fixes #139
closed 2021-01-26 by Sebazzz