CR6Community/Marlin #141: [FR] Allow 180 degree rotation of the display

rancid-racer, 2021-01-12:
I would like to be able to rotate the image 180 on the initial orientation for my CR-6 so that the SD Card Slot is at the top of the board instead of the bottom.  This allows for an opportunity to reprint a back cover and make it easier for updates without removal of the panel.

rancid-racer, 2021-01-12:
> I used a 15cm micro sd extender. It was a tight fit. I had to cut a notch in the back cover and pulled the display off of the bracket to insert the cable but it works well. I didn't have problems sticking the display back on but that could be an issue.

I purchased one as well and am currently measuring up the back cover to reprint with a notch already in place.  I just think that flipping the screen makes more sense but I'm working with what I am capable of.

Sebazzz, 2021-01-12:
I agree! I'm going to build this because I'm too cheapskate to buy an SD adapter.

ritchiedc, 2021-01-12:
Don't forget you'll still have to extend the control cable.

Thinkersbluff, 2021-01-12:
I am enjoying my external SD card slot enormously.  
Being able to upload display firmware from the main board SD or USB would be even better.

Sebazzz, 2021-01-12:
Yes, but there appear to be commands to do that, but I don't want to risk bricking the display. Besides, there is no way to show progress while the display is being flashed.

Thinkersbluff, 2021-01-12:
> currently measuring up the back cover to reprint with a notch already in place.

This may help:
https://www.thingiverse.com/thing:4693106

rancid-racer, 2021-01-12:
> > currently measuring up the back cover to reprint with a notch already in place.
> 
> This may help:
> https://www.thingiverse.com/thing:4693106

Thank you!  I got most of the measuring done but this will save me hours on Fusion360 and is exactly what I was working on.

Sebazzz, 2021-01-26:
I think that a 180 degree turn is useful - but the problem is that display is completely misaligned. This also requires then a new display mount to be printed. For this reason, until such mount is available, I'm putting this on low priority.

Sebazzz, 2021-02-02:
Closing this - feel free to reopen when a mount becomes available.

closed 2021-02-02 by Sebazzz