CR6Community/Marlin #142: Bed level speed

GTIpunk87, 2021-01-12:
Hello, 
 I'm just wondering if it's possible to speed up the leveling process

Sebazzz, 2021-01-12:
Yes, we could level at the original speed, but that would cause all the inaccuracy issues the previous and sometimes stock firmware also has.

GTIpunk87, 2021-01-12:
Ah ok, i wasn't sure if this was the max speed-accuracy ratio. I'm sure you had looked into that. Thank you!

krammerj2486, 2021-01-17:
I think bed leveling is Ok. I only do this once a while or when changing nozzle or so. The result is very good now and therefor speed low speed is no matter to me. I don't think repeating the leveling every print brings any advantage.

Sebazzz, 2021-01-26:
@GTIpunk87 If you're able to compile the firmware, please feel free to do some accuracy tests and report back. Marlin devs upstream agree however that slower leveling generally means more accurate leveling. We've seen this also with this release of 5 beta - that this has resolved many leveling issues for many users.

closed 2021-01-26 by Sebazzz