CR6Community/Marlin #143: Return Print head to safe area after bed levelling

quietbiker, 2021-01-12:
New V5 Firmware (and display updated).
Returned to Factory defaults.
Ran the auto bed level function via the display.

At the end the print head was left over the final probe area (bottom right) so accessing the bed for any reason is tricky.
As this function is done via the LCD, it may be better to clear the print bed with a G0 X0 Y235 F5000 to move the head fully left and bring the print bed forward.

Using 4.5.2 stock Creality Board.

Sebazzz, 2021-01-12:
We should probably run a G27 to park the nozzle, or do a G28XY.

nicwilson58, 2021-01-17:
I made a small change to the config file to make it do this.  Z is moved to a safe height and XY are homed.

Sebazzz, 2021-01-22:
This has now been fixed and will be included into the next release. Before the next release will be a fact, prerelease builds will be published in the Discord #prerelease-testing channel - so you can watch that channel for  those binaries, play already with some of the newer features. Alternatively you can [compile the latest firmware yourself](https://damsteen.nl/blog/2021/01/08/how-to-compile-cr6community-marlin-with-vscode-platformio).

closed 2021-01-22 by Sebazzz

Thinkersbluff, 2021-01-28:
"...accessing the bed for any reason is tricky."

I find the bed no easier to access if the nozzle is on the left or on the right, while the bed is at Y=0.0.
I like the idea of parking the head at the back left corner, if there is any need to access the bed after leveling. (e.g. like picking off bits of PLA because it is in my workflow to select Pre-Heat PLA or Pre-Heat PETG to get the system up to my intended printing temperatures before running ABL...

commit 5b13c961 (2021-01-22): Move probe after leveling. Fixes #143