CR6Community/Marlin #146: [FR] Z Steppers Auto-Alignment

Virgun, 2021-01-18:
I found interesting function (G34) in marlin firmware. I think G34 (which is Z Steppers Auto-Alignment) would be really cool feature to add. Instead of fighting with setting the gantry level you can just remove the z-axis belt and run G34. 

In current release G34 results in unknown command, but I think it wouldn't be too hard to implement it (like they did with CR-10S pro).

InsanityAutomation, 2021-01-18:
10S Pro doesnt, unless the board is changed to something with dual z stepper drivers there is no way to do it. You can implement the prusa crash method if you solder wires to get UART to reduce stepper current though.

Sebazzz, 2021-01-18:
Correct - without extensive hardware modifications this is not possible, sorry. There is a single driver for two stepper motors.

closed 2021-01-18 by Sebazzz