CR6Community/Marlin #147: LCD is stacked on splash screen.

vojjta, 2021-01-18:
### Description

Hello, yesterday i flashed my cr6se for first time with community firmware and i think it is great! Thank to all contributors!

Yesterday was everything working fine after fresh firmware flash. But today i turned on printer and display keeps stocked on booting screen.

### Steps to Reproduce

1.Turn on printer

**Expected behavior:** 

LCD should boot normaly to main screen.

**Actual behavior:** 

LCD is stacked on splash screen.

**Logging:**

Recv: FIRMWARE_NAME:Marlin Creality 3D SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR-6 SE EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M155 S2
Recv: echo:M155 S2

### Additional Information

My motherboard is 4.5.2 and I flashed printer with: CF5Beta-ProbeHeaterOn-cr6-se-v4.5.2-mb-2020-12-30-14-24.zip. Cr6Se is in stock condition.

[image]

Sebazzz, 2021-01-18:
Try reflashing your firmware .bin file - rename it before flashing. You may need to flash stock or alpha 4 firmware first to give the bootloader a "nudge". You have a v4.5.2 board and any firmware seems to be able to get corrupted there. Otherwise  you can also flash the prerelease from the Discord.

vojjta, 2021-01-18:
After flashing alpha 4 printer had same issue.

Problem was solved when I flashed back stock CR-6 SE-V2.0.1.2 and than CF5Beta again.

Maybe problem is in SD Card I used. SD Card used for flashing LCD seems is not propably working for flashing motherboard becouse i did`t notices small pause on the start of sceen. 

Now everything is working as it should bul i will observe it for few next days if it fall back into tha same problem...

Sebazzz, 2021-01-18:
For what it's worth, the start screen is fake. The progress bar is fake. It just runs a counter to show the progress bar (also in stock firmware 😅) and then navigates to the main screen. So there isn't any kind of complex logic at the "end" of the progress bar. Pure firmware corruption in this case.

closed 2021-01-18 by Sebazzz

vojjta, 2021-01-18:
:smile: it is good to knew that :smile: 

Thank you for your time and answers.