CR6Community/Marlin #148: [FR] Display bed levelling results numerically

tangcla, 2021-01-18:
On my previous printer, it had custom firmware (ADVi3++ for the Wanhao Duplicator i3 Plus / Cocoon Create Touch) and after using the BLTouch bed levelling, it would show the results of the bed mesh (see screenshot below). Is it possible for something like this to be implemented? Useful for those not brave enough to plug in an Raspberry Pi into the USB port :)
[image]

Sebazzz, 2021-01-18:
I think that #145 by @InsanityAutomation handles this. @grobux and I need to take a look at that.

Sebazzz, 2021-02-01:
We're trying to get red/green/blue colors working like in a bed visualizer, but DWIN is a bit of a pain in the back 😅

[image]

InsanityAutomation, 2021-02-01:
I still recommend setting it to odd numbers in the grid. 5x5 gets the center home point in so a large deviation tells you there is a repeatability issue and is also highlighting the most likely point for a bowl shaped bed that would be farthest off.

Sebazzz, 2021-02-01:
I don't think that at the current bed leveling speed a 5x5 grid is acceptable for users. 🙂

InsanityAutomation, 2021-02-01:
Ive always held the philosophy get a good mesh, store it to eeprom, and just recall it. If it takes an hour who cares if you only do it once a quarter. Things shouldnt change frequently on these.

Sebazzz, 2021-02-01:
I know that, you know that, but this firmware is also used by people who rather level before each print.

InsanityAutomation, 2021-02-01:
Right, I try to take the complaints from them as an opportunity to educate them as to why. Im not willing to sacrifice that much accuracy and the best sanity check for that use case. I will note however than im also fine with 3x3 on beds this small and rigid, its getting that center point that I am concerned with. The 300mm machines I push more 5x5 and 400 7x7, 500 11x11.

Thinkersbluff, 2021-02-01:
Some users may change nozzles or clear out nozzle clogs. 
Some change print platforms from flex to glass and back. 
Some take the platform off to wash it and don’t trust it to be perfectly flat or in the same place when put back.  
There are many rational use cases wherein ABL is more frequent than monthly.

Patience is definitely a useful virtue when owning and operating a 3D printer, no argument there.  Some of us also feel the need for speed, and not only when printing...

Sebazzz, 2021-02-01:
> Some users may change nozzles or clear out nozzle clogs.

Allow me to pedantic for a moment: nozzle changes do _not_ influence bed leveling because the nozzle effectively _is_ the probe. That a nozzle has more height or is lower does not matter. 

> I still recommend setting it to odd numbers in the grid. 5x5 gets the center home point in so a large deviation tells you there is a repeatability issue

I bet that is exactly the reason why Creality didn't enable a 5x5 grid - because you'd see that (3,3) is not always zero.

Thinkersbluff, 2021-02-01:
> nozzle changes do _not_ influence bed leveling because the nozzle effectively _is_ the probe. That a nozzle has more height or is lower does not matter.

Good point.  Shows how accustomed I have become to life with my Ender 3+ BLTouch - I have gotten into a pattern of “best practices” while forgetting from whence they came.  Thanks. Maybe I will be able to stop myself doing that on the CR-6...

Sebazzz, 2021-02-02:
I have another argument for keeping it at 4x4: Not all the kinks in bed leveling have been fully worked out, and although CF5Beta is a great success for many people, for some people the bed leveling is still working barely. I'd rather not introduce additional room for inaccuracies.

Thinkersbluff, 2021-02-03:
Congrats on finding a way.  
It would be kind to make clear whether bottom left on the screen = front left on the bed ( the OP’s photo suggests one easy way.)
I am already wondering what red means and whether the colour thresholds will be user modifiable. (Sorry - it’s in my nature...)

tangcla, 2021-02-03:
> Congrats on finding a way.
> It would be kind to make clear whether bottom left on the screen = front left on the bed ( the OP’s photo suggests one easy way.)
As devil's advocate, the bed levelling mesh (points 1-16) doesn't show either... but you would assume that it is the same orientation :)

Thinkersbluff, 2021-02-03:
> you would assume that it is the same orientation :)

Yes I would.  

Speaking as a guy who still can’t tell which oven knob turns on which burner based on those ubiquitous icons they put beside each knob, though, I prefer to advocate for UI designs that don’t make me guess.  ;)

Thinkersbluff, 2021-02-08:
Just for the record - we had a very good exchange and made some good  progress on the Discord LCD-Discussion channel on 7 Feb. 

https://discord.com/channels/759603746270609428/804078252661800981/808001956319592520

Thinkersbluff, 2021-02-11:
I keep thinking the only reason we display these measures to 3 decimal places is that others do.  If the sensor itself is only repeatably accurate to 3 decimal places, surely the results should only be reported to 2 decimal places...  Maybe even the MAX grid would fit, then.

Sebazzz, 2021-02-14:
[image]

Leveling visualization for the MAX has been added as well.

Sebazzz, 2021-02-14:
With the work for the CR-6 MAX also completed, I'Il close this as this has been fully implemented. Please suggest any improvements in new issues.

closed 2021-02-14 by Sebazzz