CR6Community/Marlin #150: Heater Timeout between touchscreen and octoprint

dwendooley, 2021-01-19:
### Description

Starting a print using octoprint does not stop the timer for manually-set hotend heating timeout.

### Steps to Reproduce
1. Design/download model and slice it.
2. Start up printer and Pi. Preheat bed and hot end so you can load filament as appropriate.
3. go to slicer and send to octoprint
4. go into octoprint and start the print
5. normal sequence will start, but, depending on timing, the hot end will timer will time out and zero the hot end during the printing process, obviously to bad effect.
6. clear the bed and start the print again from octoprint only.

**Expected behavior:** 
Setting the temp in the start g-code M105/M109 ? ( I think) should stop/reset the hot end timer I would think.

**Actual behavior:** 

Hot end timer times out and hot end gets too cold for PLA during octoprint prints.

**Logging:**

Printing at the moment, can update later.
### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

Stock mainboard first version.

I did use the correct firmware version, I had double-checked with Seb on Facebook.

Sebazzz, 2021-01-19:
Thank you for your report. I think this might be fixed already on the latest sources.

Two things you can try:
- In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try the bugfix ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)).
- Otherwise, put the temperature commands [after the G28 command like this](https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping) until an official release is made.

Sebazzz, 2021-01-22:
Assuming this is fixed I will close this bug now - if you find it isn't fixed @dwendooley please feel free to reopen.

closed 2021-01-22 by Sebazzz