CR6Community/Marlin #151: SD CARD is only detected after power cycle

IckZ, 2021-01-21:
Hello,
today I've installed the btt skr cr6 Mainboard and flashed both (mb and stock display) with the Marlin-2.0.7.3-cr6-community-release-5-beta firmware. 

Now I have the problem that the printer only read the sd card if I power off and power on the printer. If I want to change the sd card I have to power cylce again to see the content of the sd card. If I only remove the card the content unter "print" does not refresh or gets empty as it should.

Do you have any idea why this happens?

Sebazzz, 2021-01-21:
Please follow the issue template, have you collected a log?

IckZ, 2021-01-21:
Sorry I do not have the possibilty to log because I do not use for example octoprint. I can reproduce this error as follows:

1. Start the printer (no sd card inserted)
2. Insert a SD Card#1with gcode files in root
3. Click on print and see all files listed
4. remove SD Card 
5. Click on print an still see all files listed from SD Card#1
6. Insert SD Card#2
7. Click on print and still see all files listed from SD Card#1
8. reboot the printer
9. Click on print and see all files listed from SD Card#2
10. Loop from 2.

IckZ, 2021-01-22:
Hello,
I've flashed the motherboard and the stock screen again but this time with the original btt firmware and the crealtiy screen firmware 1.0.2 and my problem is solved. So it seems to be a problem in the community firmware. I'm sorry that I cannot provide a log.

https://github.com/bigtreetech/BIGTREETECH-SKR-CR6/tree/master/firmware/firmware(CR6%20Original%20Touch%20Screen) from the 3.12.2020 AND CR-6 SE_32-bit_V1.0.2.bin_200907.rar (https://www.creality.com/download)

Sebazzz, 2021-01-22:
This has now been fixed and will be included into the next release. Before the next release will be a fact, prerelease testing will occur in the Discord #prerelease-testing channel - so you can watch that channel for earlier binaries or [compile the latest firmware yourself](https://damsteen.nl/blog/2021/01/08/how-to-compile-cr6community-marlin-with-vscode-platformio).

closed 2021-01-22 by Sebazzz