CR6Community/Marlin #152: Changing Z Offset resets bed and extruder temp to 0

ThisIsMyUniqueUsername, 2021-01-24:
### Description

Changing Z-Offset in LEVEL menu and exiting back to the home screen causes set extruder and bed temperatures to go back to 0. This is especially annoying when trying to fine-tune the strain gauge sensitivity or level X axis.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Set a desired bed and extruder temperature - either manually or using "Preheat XXX" saved template
2. Go to LEVEL menu, change Z Offset up or down and tap BACK button on the top of the screen
3. The display is now showing home menu with "Settings Stored" message and both temperatures set to 0'C.

**Expected behavior:** 

I would expect the set temperatures to stay the same.

**Actual behavior:** 

Both extruder and bed temperatures are set to 0 and the printer starts cooling down.

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
[image]
[image]
[image]
[image]

Sebazzz, 2021-01-24:
This is kind-of by design. After leveling you usually want to maintain temperature because you're setting the Z-offset. After setting the Z-offset you're either done or you go printing.

ThisIsMyUniqueUsername, 2021-01-24:
> This is kind-of by design. After leveling you usually want to maintain temperature because you're setting the Z-offset. After setting the Z-offset you're either done or you go printing.

If I'm done then I am likely to turn off the printer, but when I want to print after that, it causes unnecessary delay before the nozzle heats up again.

I've just spent an evening trying to diagnose ABL not compensating correctly for the bed with back corners printing way too high. Half of this time is just waiting for things to preheat.

Sebazzz, 2021-01-25:
I understand, but I'm not sure keeping the temperatures set is a good option either.

Damned if you do, damned if you don't. @Thinkersbluff Any opinion?

ThisIsMyUniqueUsername, 2021-01-25:
> I understand, but I'm not sure keeping the temperatures set is a good option either.
> 
> Damned if you do, damned if you don't. @Thinkersbluff Any opinion?

I realise it's quite a niche use for the printer and not something people do often.

Thinkersbluff, 2021-01-25:
> @Thinkersbluff Any opinion?

It sounds as though the “ask” is to support a workflow in which the user is having problems levelling their bed.
If I understand well, the workflow currently used by @ThisIsMyUniqueUsername is causing the heaters to go off when they exit the Level screen and that by the time they have turned those heaters back on, it takes “too long” for the system to heat up again when they are ready to print a test pattern.

It might help if @ThisIsMyUniqueUsername describes the entire workflow. There may be alternative ways to use the existing design  (e.g. “tap to modify”) which would mitigate the issue without a design change. 
A better understanding of the workflow might also suggest new ideas (e.g. adding a TBD “Print Test Pattern” option to the Level menu)

Thinkersbluff, 2021-01-25:
> Damned if you do, damned if you don't. @Thinkersbluff Any opinion?

And damned if you’re not consistent.   

If I select Prepare->Filament Extruder->Extrude, the printer heats the nozzle to 200C.  If I select “Back” at the end of that process, the nozzle temperature target remains at 200.  If the design is intended to reset target temperatures to 0 when returning to the Main menu, unless they have been set deliberately set higher using the Prepare->PreHeat functions, they should also return to zero after returning from the Filament Extruder screen.

Since the Pre-Heat commands are designed to be an exception to that “rule”, it seems reasonable to question why the pre-heat settings are lost when returning to Main from ABL.

closed 2021-01-26 by Sebazzz

Sebazzz, 2021-01-26:
I leave the heaters on since we have the hotend idle timeout anyway. With this I consider the issue resolved.

commit 4d73c0f1 (2021-01-26): Keep heaters on when returning from the leveling menu. Fixes #152

We have the hotend idle timeout now enabled anyway so the printer is safe.