CR6Community/Marlin #153: ABL process doesn't end until BACK button is pressed - not sure if bug or design choice?

ThisIsMyUniqueUsername, 2021-01-25:
### Description

When running ABL, BACK button needs to be pressed on the screen tor the process to complete, otherwise the printer won't accept anything from Octoprint

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Go to ABL and start the process
2. Wait for the ABL to finish probing all steps
3. Try controlling printer from Octoprint - nothing happens and "busy: processing" message repeats in terminal
4. Click BACK button on the printer screen.

**Expected behavior:** 

I expected the mesh values to be saved and printer return to ready state at the end of the process

**Actual behavior:** 

Printer does not respond to any Octoprint input.

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
[image]
[image]

Sebazzz, 2021-01-25:
Please check the log: It says "Waiting to heat up before continueing". 

It didn't wait for you to press "back", it was simply getting back up to temperature.

ThisIsMyUniqueUsername, 2021-01-25:
> Please check the log: It says "Waiting to heat up before continueing".
> 
> It didn't wait for you to press "back", it was simply getting back up to temperature.

The first time I tried it it sat there for at least 10 minutes. The second time it started responding as soon as I pressed BACK, see the second screenshot. 

I will try leaving it for longer again after my current print ends just to confirm.

Thinkersbluff, 2021-01-25:
> I will try leaving it for longer again after my current print ends just to confirm

Probably best to not suppress temperature reports in the log, in this case.

Sebazzz, 2021-01-25:
Yes, good point.

Thinkersbluff, 2021-01-25:
On my printer, running the pre-release CF6, I am able to print from Octoprint while on the Level menu screen, even if I initiate the print command from Octoprint while the ABL function is still restoring nozzle temperature to 145, but the print itself only begins after the G29 command has finished restoring that temperature.  The screen itself yields no obvious clues.  

This may be a case where M117 messages might provide helpful feedback, if they were being displayed?

Sebazzz, 2021-01-26:
> This may be a case where M117 messages might provide helpful feedback, if they were being displayed?

A sudden commit appears. 

---

With this I consider the issue resolved - watch the #prerelease-testing channel for an earlier build if this is blocking to you. If you are able to reproduce that it keeps hanging feel free to reopen this bug.

closed 2021-01-26 by Sebazzz