CR6Community/Marlin #154: Cancelling print sometimes moves nozzle to a corner and mid again, extruding filament

ThisIsMyUniqueUsername, 2021-01-25:
### Description

I rarely cancel prints in the middle but recently I've been testing some things and I came across this bug when printing from Octoprint.

Normally stopping a print from the printer screen moves the head to corner and bed forward. Sometimes randomly (one out of 5, maybe 8 prints) the head moves to the corner but immediately moves back to the middle of the bed. Sometimes slowly, sometimes quicker, once it even extruded filament fast enough to grind it in the extruder gears. I observed this issue 4 times so far.

See Terminal output and a photo (printed model already removed from the bed)

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Start a print job, wait for the printer to home, preheat and start printing
2. Stop the print using STOP button on the main screen
3. Print head moves to the corner, then back to the middle again at various speeds, sometimes extruding filament on the way. This only happens once every few prints, no pattern found so far.

**Expected behavior:** 

I expect the head to stay in the corner and initiate cooldown

**Actual behavior:** 

Print head moves to the corner, then back to the middle again at various speeds, sometimes extruding filament.

**Logging:**
here)

(print job goes 
Send: N8456 G1 X113.548 Y119.991 E304.83812*88
Recv: ok
Send: N8457 G1 X113.398 Y120.176 E304.84604*87
Recv: ok
Send: N8458 G1 X113.176 Y120.363 E304.85569*85
Recv: ok
Send: N8459 G1 X112.856 Y120.509 E304.86739*80
Recv: Invoking handler for screen 40with VP=4104 value=2
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Send: N8460 M108*16
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: SetNewScreen: 36
Recv: SetNewScreen (saving): 36
Recv: //action:notification Print Aborted
Recv: SetNewScreen: 28
Recv: SetNewScreen (saving): 28
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: X:5.00 Y:225.00 Z:20.00 E:0.00 Count X:400 Y:18000 Z:8000
Recv: ok
Recv: ok
Recv: ok
Send: N8461 M84*36
Recv: ok
Send: N8462 M104 T0 S0*25
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: ok
Send: N8463 M140 S0*92
Recv: //action:notification CR-6 Community Ready.
Recv: ok
Recv: //action:notification CR-6 Community Ready.
Send: N8464 M106 S0*89
Recv: ok
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
[...]
Send: M876 S0
Recv: M876 Responding PROMPT_GCODE_INFO
Recv: ok
[...]
Recv: Invoking handler for screen 28with VP=4098 value=2
Recv: SetNewScreen: 42
Recv: SetNewScreen (saving): 42
[...]
Recv: Invoking handler for screen 42with VP=4144 value=5
[...]
Recv: Invoking handler for screen 42with VP=4158 value=9
Recv: SetNewScreen: 28
Recv: SetNewScreen (saving): 28
[...]
Recv: Invoking handler for screen 28with VP=4098 value=2
Recv: SetNewScreen: 42
Recv: SetNewScreen (saving): 42
Recv: Invoking handler for screen 42with VP=4146 value=1
[...]
Recv: Invoking handler for screen 42with VP=4144 value=5
[...]
Recv: Invoking handler for screen 42with VP=4158 value=9
Recv: SetNewScreen: 28
Recv: SetNewScreen (saving): 28
[...]
Recv: Invoking handler for screen 28with VP=4098 value=2
Recv: SetNewScreen: 42
Recv: SetNewScreen (saving): 42
Recv: Invoking handler for screen 42with VP=4146 value=1
[...]
Recv: Invoking handler for screen 42with VP=4158 value=3
Recv: SetNewScreen: 43
Recv: SetNewScreen (saving): 43
[...]
Recv: Invoking handler for screen 43with VP=4182 value=3
Recv: SetNewScreen: 42
Recv: SetNewScreen (saving): 42
Recv: Invoking handler for screen 42with VP=4158 value=9
Recv: SetNewScreen: 28
Recv: SetNewScreen (saving): 28
[...]
[image]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Sebazzz, 2021-01-25:
I've also noticed one or two times. I've not been able to track this one down yet. It seems only to happen when printing from a host.

ThisIsMyUniqueUsername, 2021-01-25:
> I've also noticed one or two times. I've not been able to track this one down yet. It seems only to happen when printing from a host.

I'm glad it's not just my printer, hope you find the root cause. Please let me know if I can be of more help.

Sebazzz, 2021-01-26:
This appears to be an upstream bug. As you say, it doesn't happen often.

Sebazzz, 2021-01-26:
I've done an attempt to fix this by clearing the command queue as well in commit 5534e4ead4719060175c2a5e673dae8f554c1e60 - if this persists I suggest you file this upstream, that cancelling a print `ui.cancel_print` still processes a few commands. I'm not sure how resolvable this is - this might just be a race condition.

A prerelease build will soon be published on the #prerelease-testing Discord channel.

closed 2021-01-26 by Sebazzz

Thinkersbluff, 2021-02-05:
@Sebazzz - I had one of these "unholy moments" tonight, with the printer (or maybe Octoprint) trying hard to "keep going" after I stopped a print.  The attached file captures the key details from Octoprint terminal.  The key words may be, "Did not receive parseable position data from printer within 10.0s, continuing without it"
[nono.txt](https://github.com/CR6Community/Marlin/files/5933251/nono.txt)

I am repeating my feedback here to keep all the info on this issue in one place.

Your response suggests this may be an upstream bug related to https://github.com/MarlinFirmware/Marlin/pull/13547, is that right?

Thinkersbluff, 2021-02-05:
So - to repeat my comment in the #Octoprint channel, in case others investigate this issue via GitHub:

I went into the Octoprint Serial Connection Behaviour settings and unchecked both of the Cancelling behaviour options.

I suspect the message,  "Did not receive parseable position data from printer within 10.0s, continuing without it" is coming from Octoprint itself, because I had checked the first option and I wonder whether the M108 sent by the second option might sometimes compromise the Stop processing in a way that causes the print to try to continue.   All guesses at this point, since some quick testing failed to duplicate the issue.

[image]