CR6Community/Marlin #159: Community Firmware vNext provisional release notes

Sebazzz, 2021-01-27:
Quick draw-up. Still need to determine if we're going to repeat everything from beta 5 too.

## Features

### All hardware configurations
Allow you to configure if you want to trade better bed leveling and Z-offset accuracy for speed, and allow you to set the preheat nozzle and bed temperature (issue #106 )

[image]
[image]
[image]

Allow overriding slicer retractions and real-time retraction tuning (issue #157) 
*todo add documentation link*

[image]

[image]

## CR-6 stock touch screen

Add linear advance tuning (issue #158)

[image]

Add hot-end PID tuning (issue #60)

[image]

Add interactive e-step calibration (issue #57)

[image]

[image]

[image]

## Enhancements

- Return nozzle to safe area after leveling (issue #143)
- Enable usage of imperial (inch) units (issue #149)
- First heat the nozzle and bed before resuming from power loss recovery (issue #149)
- Enable [Octoprint Meatpack](https://github.com/scottmudge/OctoPrint-MeatPack/) support
- Set hot-end heater timeout from 5 to 30 minutes (issue #116 and issue #150)
- Add progress to filament load/unload menu (issue #125)

### CR-6 stock touch screen only

- Don't cool down when leaving the leveling menu (issue #152)
- Document touch screen configuration flag in Configuration.h (issue #139)

## Issues resolved

- Rapid Z-offset adjustment causes factory reset (issuer #123)
- Units are displayed as mm but are actually in centimeters (issuer #111, #118m #129)
- Screen sound and stand-by settings not immediately activated after boot (#119 and #113)
- Cancelling a print sometimes processes a few commands before truly ending the print (issue #154)
- Bed temperature is sometimes lowered after homing (issue #132)

### BTT SKR CR-6 board in combination with CR-6 stock touchscreen only

- SD card file list not refreshed when reinserting SD card (issue #151)

For a [full list of added features and bugs fixed - see the Community Firmware Release 6 milestone](https://github.com/CR6Community/Marlin/milestone/3).

Thinkersbluff, 2021-01-30:
@Sebazzz - If it is now possible to make “Probe with Heaters Off” configurable from the new ABL settings menu, could this be an option on the Control menu, impacting all probe operations? Or is this a neat trick that only affects ABL?

Sebazzz, 2021-01-30:
It involves all probe operations, so homing too, but there is no space in the control menu until we take a serious look at #121

Thinkersbluff, 2021-01-30:
> there is no space in the control menu until we take a serious look at #121

Understood.  I have some ideas on that score, too.

Sebazzz, 2021-02-13:
Updated the release notes. Included a part of the beta 5 release notes as well.

Thinkersbluff, 2021-02-14:
Some users may appreciate a pointer to this CNC Kitchen Youtube video, to help them make sense of the advice, "Try playing with the maximum resolution and maximum deviation settings in your slicing profile." 

https://youtu.be/Hvw3DrVAeTA

Thinkersbluff, 2021-02-14:
"This release of the touch screen firmware comes with a new version of DGUS OS (the underlying operating system of the touch screen) to fix text rendering issues." 

Did this get rolled back?  I don't see any change to the OS filenames.
Ref https://github.com/CR6Community/CR-6-touchscreen/issues/30

Sebazzz, 2021-02-14:
> "This release of the touch screen firmware comes with a new version of DGUS OS (the underlying operating system of the touch screen) to fix text rendering issues."
>
>
>
> Did this get rolled back?  I [...]

Compared to the stock Creality firmware. I shall clarify it.

Thinkersbluff, 2021-03-04:
I find the "Restore" defaults a bit confusing on the retraction override screen.

- Why 8mm/s default Restore speed instead of the same speed as Retraction? (Makes it impossible to set a matching speed + makes the head pause each time it comes back down to the print, while the nozzle pressure is restored)
- I hesitate each time I see Restore distance of 0mm, thinking, "if the Restore speed controls bringing the filament  back into the nozzle, might the filament stay retracted if I have a value of zero here?"
- 
Maybe just my funny brain...

Sebazzz, 2021-03-07:
> makes the head pause each time it comes back down to the print, while the nozzle pressure is restored

Sorry, that is a feature (Marlin feature that is).

> Makes it impossible to set a matching speed

Why?

> I hesitate each time I see Restore distance of 0mm, thinking, "if the Restore speed controls bringing the filament back into the nozzle, might the filament stay retracted if I have a value of zero here?"

What do you propose? It is the same settings as these slicer settings:

[image]

[image]

Thinkersbluff, 2021-03-07:
> > Makes it impossible to set a matching speed
> 
> Why?

On my system, these speeds only change in increments of 5mm/sec. With an initial difference of 8mm/sec between the two, I can not make them match each other. 
I don’t understand why a default value is a “feature”.  
It would make more sense to me if the “Restore speed” was labelled “Prime Speed” and if it was defaulted to the same value as the Retraction speed, because I only use Cura.  

> What do you propose? It is the same settings as these slicer settings:

It is unfortunate that all slicers do not standardize on the same terminology.  If the label was “Extra Prime Amount” I think it would be more clear what will happen with a non-zero value.  

Selfishly, perhaps, I recommend standardizing on Cura terminology and default settings when implementing “slicer override” settings.  At least that way, users who use Cura are better able to correctly guess what each control does, the release notes can explain the equivalent settings in other popular slicers, like S3D, PrusaSlicer, Slic3r, SuperSlicer, etc..

Sebazzz, 2021-03-08:
@Thinkersbluff Please open a separate issue for clarifying the terminology. This needs some graphics work.

Thinkersbluff, 2021-03-08:
> @Thinkersbluff Please open a separate issue for clarifying the terminology. This needs some graphics work.

Done
https://github.com/CR6Community/Marlin/issues/213

Thinkersbluff, 2021-03-11:
Here are some updated screenshots, from CF6Pre5.
[Screenshots.zip](https://github.com/CR6Community/Marlin/files/6123875/Screenshots.zip)

siduenho, 2021-03-13:
Just a small hint regarding #208 

Would you mind to add a note, that this feature will be usable with Cura Release 4.9, too?
This seems to be implemented in https://github.com/Ultimaker/Cura/pull/8403

This may help people like me to not start on implementing this for Cura because it will be available soon 😆

Sebazzz, 2021-03-16:
Check, I've added a note!

closed 2021-03-27 by Sebazzz