CR6Community/Marlin #160: CF6-Prel2 View Mesh Displays Screen 52 instead + "Back" does not work

Thinkersbluff, 2021-01-28:
[RunABLNoViewMesh.log](https://github.com/CR6Community/Marlin/files/5884475/RunABLNoViewMesh.log)

-->

### Description

The View Mesh function displays screen 52 instead of displaying mesh values

### Steps to Reproduce

1. Flash CF6-Prel2 firmware to MB and Display
2. Reset Factory Values
3. Select Level -> Run ABL
4. Wait for leveling process to probe all 16 points
5. Wait for printer to return nozzle to left front corner
6. Check Terminal, to review the reported mesh values
7. Select View Mesh Values
8. Printer displays ABL screen 52. 
9. Back button beeps but does not exit the displayed screen.

**Expected behavior:** 

IF View Mesh is not yet implemented, button should be suppressed.
IF View Mesh is supposed to work, it should be referenced in the release notes + it should work

**Actual behavior:** 

See attached log.
(NOTE: The log also seems to show a problem on the serial interface, which may be a factor?)

**Logging:**

See attached file.

### Additional Information

CR-6 SE, 4.5.2 Motherboard.

[Which *exact* firmware version did you flash? ]
CF6-Pre2-cr6-se-v4.5.2-mb-2021-01-27-23-23.zip

[Provide pictures or links to videos that clearly demonstrate the issue.]
[image]

Same issue I reported before this release, in Discord: https://discord.com/channels/759603746270609428/791082544899555358/803676682577379328

Sebazzz, 2021-01-28:
No I should have mentioned that in the testing instructions - the mesh screen is being worked on by @grobux 😅

Sebazzz, 2021-01-31:
Closing this as the back button works now and the rest is tracked in #148

closed 2021-01-31 by Sebazzz