CR6Community/Marlin #161: Different height between stock firmware and community firmware

gmueller2000privat, 2021-01-29:
### Description

Not sure if it's a bug or something else but I had to use 0.20 als z offset with the stock firmware for PLA and now with the community firmware I have to use 0.10. 

### Steps to Reproduce

Use stock firmware, level the bed and afterwards switch to community version and compare

**Expected behavior:** 

I would expect that I have to use the same heigth

**Actual behavior:** 

0.10 difference while leveling.

**Logging:**

### Additional Information

Used CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

siduenho, 2021-01-30:
I can confirm this behaviour. But I think this is expected as noted in the release notes of alpha 4:

https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-4-alpha

> More often taring/zeroing the strain gauge probe, reducing possible issues with the bowden tube or cable pulling the strain gauge up and causing false or inaccurate readings (issue #32) - for this reason you may need a lower Z-offset than in previous release. You may also be able to increase your strain gauge sensitivity to allow less flexing of the hot-end while probing.

Sebazzz, 2021-01-30:
This is not a bug, but expected behavior.

closed 2021-01-30 by Sebazzz