CR6Community/Marlin #162: Filament retract command preheats to 200C, can't change temperature, can't cancel, can't see status

jurkstas, 2021-01-29:
### Description

Retraction automatically preheats nozzle. That's awesome. But it preheats the nozzle to 200C and if I try to change that to 240C, it immediately  changes back to 200C and proceeds to retract when it reaches the temperature. Also,I don't know if I can cancel the retract operation before it happens nor if it is doing anything other than heating. I see it as User Interface issue, mostly, because I _can_ preheat the nozzle to proper temperature _before_ I hit retract and it _does_ what it is supposed to do, but _without_ feedback on screen.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Prepare -> FILAMENT EXTRUDER
2. Press "Retract"
3. Observe END temperature

**Expected behavior:** 

1. either be told to preheat or get to choose temperature
2. see status (preheating -> moving Z up -> yanking? -> advancing a bit? -> retracting -> moving Z down)
3. be able to cancel operation because it has so much going on

**Actual behavior:** 

No user feedback other than heating to 200C and moving Z and Extruder

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information
Model: CR-6 SE
Mainboard revision: 4.5.2 
Firmware: [CF5Beta-ProbeHeaterOn-cr6-se-v4.5.2-mb-2020-12-30-14-24](https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-ProbeHeaterOn-cr6-se-v4.5.2-mb-2020-12-30-14-24.zip)
No intentional setting changes, no mods.

siduenho, 2021-01-30:
I think this is standard Marlin behaviour. Because you would most probably damage your hot end by just retracting while still cold.

You can find a discussion about the naming of the menu entry in #125 (Retract/Extrude Vs Change Filament).

Apart from that: I too think that some kind of info on the screen may be a good idea. But I would like to see this for the more cautious user: so he knows that he does not need to preheat manually and can just press the button. Maybe a small info text on that screen would be a good idea?

Sebazzz, 2021-01-30:
We display some status information from CF6 Pre 2 onwards. It is however currently not possible to accept commands from the touchscreen while this synchronous operation is being performed - this would require some refactoring.

Thinkersbluff, 2021-01-31:
@Sebazzz - The stock Marlin "Change Filament" menu allows me to declare whether to preheat the nozzle for Type 1 (PLA) or Type 2 (PETG), before I select Load or Unload.   It does, annoyingly, assume that if I unload PLA, then it should preheat to PLA temperature when I select "Load" & sometimes I would rather be able to Unload PLA but Load PETG.  

Can you see a way to allow a user to set the Nozzle temperature to be used by the M701/M702 commands before they press "Extrude+" or "Retract-?"  I know when my filament has become stuck in the Ender 3 nozzle, I am sometimes able to heat the nozzle to 260 and then use the Unload function to save me from having to strip down the system to get the filament to let go...

(And - now that we have the esteps function - can we please revisit that naming decision & rename the "Filament Extruder" menu to "Change Filament" & rename "Retract-" to "Unload" and "Extrude+" to "Load"?)

Sebazzz, 2021-01-31:
@grobux I agree with @Thinkersbluff - we need to rename it back to "Change filament" and "Unload and Load".

grobux, 2021-01-31:
@Thinkersbluff Thanks, labels updated (https://github.com/CR6Community/CR-6-touchscreen/commit/c14dbb40ff2fbb695582ecf37ef16c36bc8d86a6)

Sebazzz, 2021-01-31:
> 
> 
> @Thinkersbluff Thanks, labels updated ([CR6Community/CR-6-touchscreen@c14dbb4](https://github.com/CR6Community/CR-6-touchscreen/commit/c14dbb40ff2fbb695582ecf37ef16c36bc8d86a6))

I think you meant to say https://github.com/CR6Community/CR-6-touchscreen/commit/3cf641233973c4cd7b994036a804169a2cabec28 😅

grobux, 2021-01-31:
Oops, yes! 😅 I linked the parent commit...

Sebazzz, 2021-02-03:
Okay with the status now visible and the ability to set the nozzle temperature beforehand, this is the best we can do for now. 😄

closed 2021-02-03 by Sebazzz