CR6Community/Marlin #167: Can't print, operation halts and printer beeps

NickDevoctomy, 2021-02-01:
__Issues that do not follow this issue template will be closed__

### Description

When trying to do a print with the "happy new year" beta edition, I can't get the print to start, with odd heating patterns, followed by a repeated beep tone.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Heat printer to 120 end and 65 bed
2. Start print

**Expected behavior:** 

1. Heating levels should be reached
2. Printing should start, which in my standard file should do the following,
- Autohome
- Level bed
- Start print 

**Actual behavior:** 

1. Heating levels reached
2. Auto home starts
3. Printer drops hotend temp to 0 whilst going down to touch bed
4. Printer sets hotend temp back to 120 when going up from bed
5. Once autohome completes it then moves to max y and x 0 position
6. Printer then attempts to heat to 170 end and 65 bed.  Not sure where it's getting 170 from, my gcode is created for 210 hotend.
7. Once temp reached printer beeps on / off repeatedly

### Additional Information

BTT SKR 6 Board with Stock creality touch screen. Release files downloaded from GitHub this morning.

NickDevoctomy, 2021-02-01:
For reference this is my gcode,

[xyzCalibration_cube.zip](https://github.com/CR6Community/Marlin/files/5904579/xyzCalibration_cube.zip)

Sebazzz, 2021-02-01:
Please collect some logs, specfically run command `M119` and you will probably see that your "E-axis" endstop AKA filament runout sensor is "open".

NickDevoctomy, 2021-02-01:
Hi Sebazz, aah okay, I removed that thing ages ago.  Thanks I'll disable it.

closed 2021-02-01 by NickDevoctomy